prusa3d/PrusaSlicer #1946: First layer number is wrong (always 2) and checking/unchecing retractions or travel boxes in preview tab changes layer number

Marcin-Szydlowski-1984, 2019-03-13:
### Version
1.41.3

### Operating system type + version
Windows 8.1 Pro 64bit

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3, firmware 3.6

### Behavior
Hello,

First layer number in "Preview" tab is always 2, no matter how many layers will be.

[image]

I purged Slic3r from my computer, reinstalled it, prepared few different models using the same default settings and still the same - first layer is 2 and I cannot slide the toggle lower. But this is only bug on interface because layer numbers in GCODE are valid (started from 0 - I'm using [layer_num] variable for this) so shift in the application seems to be 2. On "Layers" tab height is presented correctly for every layer so this is valid too.

I also noticed that issue [https://github.com/prusa3d/Slic3r/issues/1428](url) still occures. I confirm that clicking repeatedly on "Travel" checkbox combined with "Retraction" changes this number:

[image]

bubnikv, 2019-03-14:
Would you please test with the latest 1.42 alpha?
https://github.com/prusa3d/slic3r/releases

Marcin-Szydlowski-1984, 2019-03-14:
@bubnikv thank you. I've just tested it and seems to work. I also noticed huge changes in UI and options for "simple", "advanced" and "expert" modes showing appropriate settings. All those changes are new features compared to 1.41.3 version or did I omit something?

closed 2019-03-14 by Marcin-Szydlowski-1984

lukasmatena, 2019-03-14:
Yes, all these changes (and many others) are part of the 1.42 series. You did not miss any major release, the release cycle is just longer than usual because of the huge number of improvements, which often had to be done in one go. The 1.42 is still in alpha phase, which is why you can't get it on the official site, just here on Github. See change logs of all the alpha versions on the release page so you don't miss anything.

Marcin-Szydlowski-1984, 2019-03-14:
@lukasmatena Ok, thanks for info.