prusa3d/PrusaSlicer #1948: Request - modification to purge tower perimeter to single clean material

bynnek, 2019-03-14:
### Version
1.41.3

### 3D printer brand / version + firmware version (if known)
MK2.5 + MMU2 (3.6.0 & 1.0.5)

### Behavior
When using soluble supports or two materials which do not have excellent adhesion to one another, the wipe tower sometimes warps and delaminates.  An example is using PLA with PVA.  Adhesion isn't stellar and on a wipe tower in particular, you get very poor layer to layer adhesion.  If they peel up from one another or warp, this can lead to the extruder colliding with the wiper tower and either cause a layer shift or knock over the wipe tower.  

What I was hoping to get was an option to make the perimeters of the wipe tower to only one material (preferably the print material and not the support material).  If this perimeter was fully purged filament and always the same each layer, it would have excellent layer to layer adhesion and give the purge tower a lot of strength.  All the "infill" of the purge tower would still be purged material as normal.  The theory would be that if the perimeters are adhering well, it will prevent the purge tower from warping and delaminating.

bynnek, 2019-03-14:
Example of one of the delaminations when using PLA with PVA that I think could be mitigated by making sure the external perimeters were not slightly mixed material.   
[image]

TorqueBias, 2019-06-03:
I've been running into this same problem myself. I've lost 3 prints so far because the PVA lets the tower come apart easily. I can crush the tower in my hand it is so weak. Adding some kind of structure to the tower using the main material would solve this.

TorqueBias, 2019-06-03:
I had a **crazy** idea after posting this. If you place a model inside of the purge block, the tool paths intersect as if the other structure isn't there. I modeled a cube in Fusion that was open on top and bottom with a small brim around the bottom. Originally I was going to use it as a shell around the purge block but it was difficult to place correctly. Instead I scaled it as an internal structure to the purge block that just roughly fits inside the volume. It prints the model before it purges, so there isn't much pva deposited on top of the PLA structure. So far my purge block is twice as tall as my previous record without having any separated layers. I'll update you later when I see if it finishes.

TorqueBias, 2019-06-04:
Update: 
This workaround worked for me! First successful PVA print. I could still break the tower in half by squeezing it, but it wouldn't crush, and it was about twice as strong as before. I think maybe a model of an "X" that is slightly bigger than the purge tower (exending from the corners?) might work better since there was still enough PVA wiped on the cube model to weaken it substantially.

bynnek, 2019-06-06:
That's pretty clever!  I'll give that at try next time I do a print with PVA.

TorqueBias, 2019-06-15:
The X definitely works better since the legs extending out of the purge block are much stronger than the parts of the model inside the block. With a brim turned on it stuck to the build plate really well. On my failed prints the corners of the purge block came apart first, and the X holds the corners together very well.

blazer828, 2019-10-18:
Hey @bynnek would you mind to share a picture of your workaround?

I am almost pulling my hair out trying to print TPU with PVA 😬

TorqueBias, 2019-10-18:
Here is what I did. I just made an X shape in Fusion and extruded it, then place it in the middle of the purge block. I made it extra tall in this pic so you can see it better.

[image]

blazer828, 2019-10-18:
So, are you using TPU with PVA? and the support material is the TPU with PVA at interfaces?

TorqueBias, 2019-10-21:
That particular model was just PLA with PVA, but the materials don't matter much. The X in the purge block was just set to be PLA. The problem is that nothing sticks well to the PVA, so you need a structure without (much) PVA contamination outside of the purge block to hold things together.

ssill2, 2021-05-11:
I would like to add that the solution with the X does work great.  Very little flaky mess with PETG + BVOH

Before
[image]

And after
[image]

Thanks for figuring this out!

lukasmatena, 2021-06-03:
This issue should be fixed for 2.4.0 release ( f6de946dd72bd6de46612c20998ccf2b55d2a134 ). Perimeters of the wipe tower are never printed by soluble filament if a non-soluble filament is used on that layer.

I'll close the issue. Let us know if you consider the issue solved after first public 2.4 alpha is released (in several weeks probably). Thanks.

closed 2021-06-03 by lukasmatena

ssill2, 2021-06-13:
so I'm slicing with the latest master build of 2.4.0 as of this morning.  I'm trying to print a simple hilbert cube with petg and bvoh.  I'm having the same poor adhesion on the wipe tower.  I'm trying again with the X through the wipetower that worked before.  If that works I'd be very interested in possibly reopening this issue and having an option in the slicer to print such a pattern in the wipe tower to improve adhesion.