prusa3d/PrusaSlicer #1950: [Feature Request] - SLA - Color faces by orientation/face normals and "slice view" in "support edit mode"

verner33ob, 2019-03-14:
### **1. Color faces by orientation/face normals**

I think it would be very useful to color the faces from green to red depending on the orientation of the particular face.
E.g. horizontal faces and almost horizontal faces are coloured red (e.g. 90° - 70°) and up to 45° the faces gradientally become more and more yellow and finally green.

Maybe those values can be edited, so in "Support Edit Mode" it's much easier to see, which areas need supports, which need more, which need less.

### **2. "slice view" in "support edit mode"**

It would be very useful to have the same slider on the side in "supoort edit mode", as there is already available in the Preview window.

Support points in the invisible areas should then become invisible too.

That way it would be much clearer to edit supports manually, since areas with densely packed support points can be hidden and also it would be much easier to edit areas, which need inside support (like the inside top of a head of a head model with 2mm wall thickness).

lukasmatena, 2019-03-14:
Thanks for the suggestions.

1. The orientation of the normal is not the only thing that decides whether or not that part needs support. That also depends on the overall geometry of the model, thickness of various parts (thin and almost vertical parts may still need supports in order to fight stresses induced during curing, etc.) The supports not only support overhangs as with FDM printing, they must also counteract the force exerted on the model during tearing the new layer from the display. Editing the supports manually will always require some understanding of the printing process and I cannot tell how much this feature would help. I would be interested to hear other opinions.

2. This is a reasonable request and funnily enough, we are already working on it. It should work pretty much like you describe, except the slicing layer will be perpendicular to the camera view direction, not the z axis.

verner33ob, 2019-03-15:
Thanks for your quick reply.

1. Yeah i know that in SLA there's more to support than FDM and certainly it needs some understanding what you are doing, if you manually edit them, but still, so far i didn't encounter any SLA-slicing software that did a perfect job. I worked with several different printers (SLA, DLP, even top-down) and several different softwares. From hobbyist software to professional softare that jewelers and the automobile industy uses, there is almost always some manual editing required. But still, all those softwares visually show overhangs, so you can much easier see the main areas, that are predestined to fail.
Of course special features (like thin almost vertical parts) need special attention.
But yeah, let's see what other will say to it :)

(Even though in FDM printing supports also don't only need to support overhangs, they also need to hold little islands down, which try to warp up :P )

2. That sounds good. But why would you do it that way? So the slicing layer would basically move simultaneously to the camera? That way i could never look inside my model.
The problem is, that with my "inside top of the head" problem, i would have to be able to look inside. Unfortunately there were not enough supports to support it enough and i had to add some. Through the hole on the bottom, which is pretty slow work :D 

But still, thanks for giving us such a wonderful software. It's amazing to see, how intuitive and pretty damn good Slic3r already is for SLA printing, even though it's the first version for that.
For FDM more than obviuosly too :D

bubnikv, 2019-04-13:
The clipping plane for editing SLA supports will be in the next beta.

lisandro52, 2020-03-11:
Any news on the colored faces by normal orientation? I think this is one of the most useful features on other (SLA) software, like Chitubox, where you can easily see where the support would be needed.
It doesn't have to be perfect, or account for edge cases like thin walls. With a simple green-yellow-red gradient would be enough for manual editing and save a ton of time (and resin).

bubnikv, 2020-03-12:
This is something for @enricoturri1966 to look into.
We may implement the color mapping by a relatively simple vertex shader calculating the slope
sqrt(n.x^2+n.y^2)/n.z
or even just 
(n.x^2+n.y^2)/n.z
and mapping the slope throw a rainbow filled texture.

The most complex likely will be to integrate the toggle switch into the UI workflow in a reasonable way.

lisandro52, 2020-03-18:
The way other softs implement this is to show the color mapping only when you're adding supports. Would that work, @bubnikv ?

enricoturri1966, 2020-03-20:
Hi all,
we have an experimental build with a prototype of slope rendering.
We would appreciate if you could test it and give us your feedback.

Slope shading can be turned on/off by pressing [d] on the keyboard or by the new menu command View->Options->Show slope.
Thresholds for coloring can be adjusted using the sliders in the dialog which pops-up in the lower right corner of the 3D view.

https://www.dropbox.com/s/rxwct4ubg2i2dpm/PrusaSlicer-2.2.0-rc5%2B8-et-slope-shader-g5b24a0fb9-202003201029.dmg?dl=0
https://www.dropbox.com/s/ws9gaaqrhv0bjtx/PrusaSlicer-2.2.0-rc5%2B8-et-slope-shader-linux-x64-g5b24a0f-202003201046.AppImage?dl=0
https://www.dropbox.com/s/1mdkys0d5c4nj3l/PrusaSlicer-2.2.0-rc5%2B8-et-slope-shader-win32-g5b24a0fb9-202003201029.zip?dl=0
https://www.dropbox.com/s/r38qyq7b141k6p5/PrusaSlicer-2.2.0-rc5%2B8-et-slope-shader-win64-g5b24a0fb9-202003201042.zip?dl=0

verner33ob, 2020-03-20:
Works like a charm :)

lisandro52, 2020-03-24:
Works amazing, thank you!

mylife4aiurr, 2020-04-15:
Was directed here from facebook. This feature was much needed. Have the Prusa SL1 and was not using Prusaslicer for a NUMBER of reasons but this was a one of them. 
In order to place manual supports effectively the user needs to "see" what needs to be supported. - thank you.
I like that you have made it real time, in such that if I rotate the model the area's needing support change as model is rotated.

closed 2026-09-29 by prusaslicer-bot[bot]