prusa3d/PrusaSlicer #1951: SLA - Several bugs and improvements

verner33ob, 2019-03-14:
### Version
Slic3rPE-1.42.0-alpha7+win64-full-201903011809

### Operating system type + version
Windows 10 Home 64 Bit

### Behavior

### **Bug 1**

Part auto-centers on print bed size changes, but supports do not.

How to recreate the bug:
Activate "preferences" -> "auto center  "parts"
Add an object and create supports. Then change the printbed size.
The object will automatically center, but the supports won't.

Only affects 3D-Editor view. In the Preview-Window the object looks fine and the exported images are also correct.
Re-slicing doesn't change the problem. I had to go into manual support edit mode and edit at least one point. Then after slicing again, the problem was gone.

[image]
[image]

### **Bug 2**
Part auto-centers on origin changes, but supports do not.

How to recreate the bug:
Activate "preferences" -> "auto center  "parts"
Add an object and create supports. Then change the origin in the printbed window.
The object will automatically center, but the supports won't.

Only affects 3D-Editor view. In the Preview-Window the object looks fine and the exported images are also correct.
Re-slicing doesn't change the problem. I had to go into manual support edit mode and edit at least one point. Then after slicing again, the problem was gone.

[image]
[image]
[image]

First time i couldn't even export anything, since the supports went outside the printbed, so i moved it to fit inside print bed. Then auto-centering again moved the part to the center and left the supports position untouched. The preview window looked correct, but the exported images were wrong

[image]

### **Bug 3**

Wrong support point selected in manual support edit mode.

Couldn't find a way to safely recreate the bug, but it happened several times. Try to find a way to ecreate hopefully during the next days.

The image attached shows a red dot, where my mouse was and the arrow points to the point slic3r thought i wanted to select.

[image]

### **Bug 4**

Again object moved, but supports didn't, when moving the object and then switching to manual editing mode, with auto-centering selected.

How to recreate the bug:
Activate "preferences" -> "auto center  "parts"
Move your part and then switch to manual editing mode. The object will suddenly jump to center and the supports will stay where they were.

### **Bug 5**

Set bed shape window visible, when it shouldn't

How to recreate the bug:
Click on "Set..."-button to change the bed-shape. Click okay. The window closes correctly.
Now hover over the Slic3r-Taskbar-Icon and then hover over the little window that shows up. Don't click on it.
Windows will now show the Slic3r-window and the bed-shape-window, that you just closed is visible.

[image]

### **Improvement 1**

When changing the origin to the center of bed, it would be good to see positive values to the right and negative values the right or to have an option, to switch between those two varieties. That way i could move an object to precise mirrored locations when i have odd numbers of my bed-size.

### **Improvement 2**

Make the bottom-Center of imported obects the origin of that object. Or to switch between that and the actual position. I thought my object was in the air, when i saw that the origin was 0.5 in z-direction, since i realised that it was just half of my objects height.

tamasmeszaros, 2019-03-15:
The days of the "auto-center" option are numbered... :) It might not even make it to the final release. Other bugs will be addressed soon. Thank you @verner33ob for the feedback.

verner33ob, 2019-03-15:
You're welcome. Thanks for that amazing piece of software.
Luckily, all those bugs are negclectable for me as a user or easy to work around and i'm already getting great and quick results.
I'm pretty amazed and using silc3r for ages, i know you're "SLA-Slic3r" will become a dream :)

bubnikv, 2019-03-17:
Bug 3 - there was an issue with the selection for number of support points above 256. That will be rectified in the beta.

bubnikv, 2019-04-02:
Mirroring should be fixed on the master, it will be in the beta2

bubnikv, 2019-08-01:
Would you please kindly close this issue and reopen one github issue per remaining issue? It should be a rule, that one github issue relates to a single issue only. Thank you.

verner33ob, 2019-08-12:
Sure, i didn't want to flood the issues section, cuz i actually wanted to write even more issues, which i didnt. But you are right, it makes much more sense doing it like this.

closed 2019-08-12 by verner33ob