prusa3d/PrusaSlicer #1952: [feature Request] custom gcode placeholder for model maximum z height

brandstaetter, 2019-03-14:
I would like to have a new gcode (and filename) variable/placeholder:

maximum model height, or build height, or z height of the final layer.

reason: 

* It would be useful in a progress message on layer change.
* It would be useful for use on a printer that supports mixing extruders to properly define a gradient over the whole build height automatically
* It would be useful to add to a filename, similar to the estimated build time.

brandstaetter, 2019-03-14:
Workaround: https://reprap.org/forum/read.php?263,764633

fe60, 2021-04-10:
The documentation mentions `total_layer_count` as placeholder [1].
Doesn't this fulfill your requirement?

[1] https://help.prusa3d.com/en/article/list-of-placeholders_205643

brandstaetter, 2021-04-18:
the layer count does not give me the height reliably, since variable layer height is a thing...

fe60, 2021-04-19:
A dedicated placeholder or the option to declare new variables in macros would be easier to use, but you can determine the maximum model height already.
```
;Layer height: [layer_height]
;First Layer Height: [first_layer_height]
;Number of Layers: [total_layer_count]
;Max. height of object(s): {first_layer_height+layer_height*(total_layer_count-1)}
```

brandstaetter, 2021-04-19:
In my understanding this would break if I use variable layer heights, i.e. switching from 0.1 to 0.2 layers in certain layer areas to increase details...

fe60, 2021-04-21:
You are right, this doesn't work for variable layer heights. But in such a scenario the height of the final layer won't be enough for your use case(s).
You will need the height for each layer. So far there is only `layer_num`.

rtyr, 2021-04-22:
Since 2.3.0-beta2, there is a new variable `max_layer_z` available in custom g-codes. It represents the maximum print Z value over the already printed layers, including sequential print.

zenturacp, 2021-05-17:
> Since 2.3.0-beta2, there is a new variable `max_layer_z` available in custom g-codes. It represents the maximum print Z value over the already printed layers, including sequential print.

So this will actually be the height in mm of the printet object?

bubnikv, 2021-05-19:
Yes, it is the height of the tallest object. We use this placeholder to
lift up the print head at the end of the print before parking.

brandstaetter, 2021-05-27:
then I consider this request closed, thank you!

closed 2021-05-27 by brandstaetter

wmmc88, 2021-06-06:
it seems like `max_layer_z` isn't available in start gcode?

tommylux, 2021-11-20:
Doesn't seem to work. Would be good for colour mixing because Marlin only accepts the value in MM
M166 A0 Z{max_layer_z} I0 J1 T4 ; Set Mixing from Z0 to totalz from T0 to T1 as alias T4 Gradient