prusa3d/PrusaSlicer #1953: Strange Exruder 1 Behavior

cosmith71, 2019-03-14:
### Version
1.42.0 alpha 7

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
MK3 MMU2

### Behavior
Tool 1 resets to an unselected profile when changing unrelated menu items.

This problem is intermittent and not always replicable. 

Set Printer to "Original Prusa i3 MK3 MMU2".  Set Print Settings to "0.20mm QUALITY MK3"  Set all filaments to "Generic PLA MMU".
Load an STL.
Change Print Settings to "020mm SPEED MK3"
Filament 1 changes to "Prusament PLA MMU2"
Change it back to "Generic PLA MMU2"
Change filament 2 color (with the color picker).
Filament 1 changes to "Prusament PLA MMU2"

See attached video and 3MF.

[Media1.zip](https://github.com/prusa3d/Slic3r/files/2968825/Media1.zip)

YuSanka, 2019-03-15:
Thanks for your report. We know about this problem and working on it in [SPE-865](http://dev.prusa3d.com/browse/SPE-865)

cosmith71, 2019-03-15:
Great!  I've seen it for a while and I thought maybe I was the only one having the problem since I hadn't seen it reported.  It will be nice to have it fixed.

Thanks!

bubnikv, 2019-04-13:
I believe this has already been fixed in beta1. Closing. Please free to reopen if you find out that it is still broken.

closed 2019-04-13 by bubnikv