prusa3d/PrusaSlicer #1954: Show computed values

guestisp, 2019-03-15:
When using % values or values based on other settings, would be useful to show, after each value, the "computed" value.

In example: nozzle diameter set to 0.4, default extrusion width set to 0, after the extrusion width field would be useful to show 0.45 because in that case, an extrusion width of 0 means "nozzle diameter * 1.125"

This would give a better outlook on each value.

supermerill, 2019-03-19:
Some values can change depending the filament/extruder/layer.
Exemple: the width depends of the current layer height.

guestisp, 2019-03-19:
Exactly, that's why showing the computed value would be' nice as the value could depend on a value in a different page

supermerill, 2019-03-19:
No, i mean the value can change over time and so you can't know the value because they are multiple ones.

guestisp, 2019-03-19:
But for simpler situation, wont be useful ? In example, the varius speed in %

supermerill, 2019-03-19:
Yes, I think so.

Note that the values actually written inside the box isn't the final one anyway, It depends of the layer size: ,the auto-cooling can reduce the speed. Also, it depends of the max volumetric speed from the filament.

guestisp, 2019-03-19:
That's exactly what I mean. The final value is very hard to predict manually, beacuse it could depends on multiple other values. That's why an "evaluated" value wrote on the side of the field could be interesting. 

If I set 20% I have to remember tons of affecting values and, after all, i'll never know which value would use the printer. Slic3r could automate this as much as possible for a easy of use

supermerill, 2019-03-19:
Even if you set 20mm/s, the value will change.

guestisp, 2019-03-19:
In example: "Small perimeter" speed value is based on "Perimeter" speed value.
If I set `Perimeter` to 50mm/s and `Small perimeter` to 10%, next to the field would be useful to see `5mm/s`

The field name doesn't matter, as some fields are based on other fields, ad additional label would improve the usability, IMHO

bubnikv, 2019-04-13:
these automatic values are a nightmare, some of the issues @guestisp is reporting are caused by this logic. It would save us a lot of time to get rid of this logic.

guestisp, 2019-04-13:
Totally agree. Too many settings are dependant on too many other settings. It's hard to know which value is really in effect for certain parameters

closed 2026-09-29 by prusaslicer-bot[bot]