prusa3d/PrusaSlicer #1955: Crash on "Cut" after layer preview has been used

wavexx, 2019-03-15:
### Version
ef939905b198460dc1e8c8ab0661dbf2518ff4c4

### Operating system type + version
Linux (debian unstable)

### Behavior
Load one object on the plate.
Switch to layer preview.
Switch back.
Attempt a cut using the cut gizmo.

Crash.

If you do the cut before switching to layer preview, cut always works.

This has been true for several weeks now.

bubnikv, 2019-03-15:
what Slic3r version?

wavexx, 2019-03-15:
On Fri, Mar 15 2019, bubnikv wrote:
> what Slic3r version?

Current master.

wavexx, 2019-03-16:
I actually get a SIGTRAP due to a BadMatch error:

````
(slic3r-gui:8806): Gdk-ERROR **: 19:52:15.953: The program 'slic3r-gui' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadMatch (invalid parameter attributes)'.
  (Details: serial 19094 error_code 8 request_code 152 (GLX) minor_code 11)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the GDK_SYNCHRONIZE environment
   variable to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)

Thread 1 "slic3r-gui" received signal SIGTRAP, Trace/breakpoint trap.
0x00007ffff542cca5 in _g_log_abort (breakpoint=1) at ../../../glib/gmessages.c:554
554     ../../../glib/gmessages.c: No such file or directory.
(gdb) bt
#0  0x00007ffff542cca5 in _g_log_abort (breakpoint=1) at ../../../glib/gmessages.c:554
#1  0x00007ffff542f60c in g_log_writer_default (log_level=6, log_level@entry=G_LOG_LEVEL_ERROR, fields=fields@entry=0x7fffffff98b0, n_fields=n_fields@entry=6, user_data=user_data@entry=0x0)
    at ../../../glib/gmessages.c:2688
#2  0x00007ffff542d907 in g_log_structured_array (n_fields=6, fields=0x7fffffff98b0, log_level=G_LOG_LEVEL_ERROR) at ../../../glib/gmessages.c:1923
#3  0x00007ffff542d907 in g_log_structured_array (log_level=G_LOG_LEVEL_ERROR, fields=0x7fffffff98b0, n_fields=6) at ../../../glib/gmessages.c:1896
#4  0x00007ffff542e32e in g_log_structured_standard
    (log_domain=log_domain@entry=0x7ffff598d017 "Gdk", log_level=log_level@entry=G_LOG_LEVEL_ERROR, file=file@entry=0x7ffff59ab838 "../../../../../gdk/x11/gdkdisplay-x11.c", line=line@entry=0x7ffff59ab28f "2769", func=func@entry=0x7ffff59abf50 <__func__.75983> "_gdk_x11_display_error_event", message_format=message_format@entry=0x7ffff59ac2c6 "%s")
    at ../../../glib/gmessages.c:1980
#5  0x00007ffff594cd6b in _gdk_x11_display_error_event (display=display@entry=0x5555561c70e0 [GdkX11Display], error=error@entry=0x7fffffff9ef0)
    at ../../../../../gdk/x11/gdkdisplay-x11.c:2769
#6  0x00007ffff5959bc3 in gdk_x_error (error=0x7fffffff9ef0, xdisplay=0x5555561b8490) at ../../../../../gdk/x11/gdkmain-x11.c:307
#7  0x00007ffff5959bc3 in gdk_x_error (xdisplay=0x5555561b8490, error=0x7fffffff9ef0) at ../../../../../gdk/x11/gdkmain-x11.c:269
#8  0x00007ffff614511a in _XError () at /usr/lib/x86_64-linux-gnu/libX11.so.6
#9  0x00007ffff6142077 in  () at /usr/lib/x86_64-linux-gnu/libX11.so.6
#10 0x00007ffff614211d in  () at /usr/lib/x86_64-linux-gnu/libX11.so.6
#11 0x00007ffff6142a55 in _XEventsQueued () at /usr/lib/x86_64-linux-gnu/libX11.so.6
#12 0x00007ffff61457f5 in _XGetRequest () at /usr/lib/x86_64-linux-gnu/libX11.so.6
#13 0x00007ffff217e73d in XIQueryPointer () at /usr/lib/x86_64-linux-gnu/libXi.so.6
#14 0x00007ffff5945ddc in gdk_x11_device_xi2_query_state
    (device=0x5555561eb820 [GdkX11DeviceXI2], window=0x555559b1c190 [GdkX11Window], root_window=0x0, child_window=0x7fffffffa330, root_x=0x0, root_y=0x0, win_x=0x7fffffffa358, win_y=0x7fffffffa360, mask=0x7fffffffa354) at ../../../../../gdk/x11/gdkdevice-xi2.c:351
#15 0x00007ffff5963105 in gdk_window_x11_get_device_state (mask=<optimized out>, y=<optimized out>, x=<optimized out>, device=<optimized out>, window=<optimized out>)
    at ../../../../../gdk/x11/gdkwindow-x11.c:3351
#16 0x00007ffff5963105 in gdk_window_x11_get_device_state (window=<optimized out>, device=<optimized out>, x=<optimized out>, y=<optimized out>, mask=<optimized out>)
    at ../../../../../gdk/x11/gdkwindow-x11.c:3339
#17 0x00007ffff5936b74 in gdk_window_get_device_position_double
    (window=0x555559b1c190 [GdkX11Window], device=0x5555561eb820 [GdkX11DeviceXI2], x=x@entry=0x7fffffffa3b8, y=y@entry=0x7fffffffa3c0, mask=0x0) at ../../../../gdk/gdkwindow.c:5026
#18 0x00007ffff5936c8d in gdk_window_get_device_position (window=<optimized out>, device=<optimized out>, x=0x7fffffffa400, y=0x7fffffffa480, mask=<optimized out>)
    at ../../../../gdk/gdkwindow.c:5077
#19 0x00007ffff73e32ce in gtk_window_motion_notify_callback () at /usr/local/lib/libwx_gtk3u_core-3.1.so.2
#24 0x00007ffff552a91f in <emit signal ??? on instance 0x555556961120 [wxPizza]> (instance=instance@entry=0x555556961120, signal_id=<optimized out>, detail=detail@entry=0)
    at ../../../gobject/gsignal.c:3447
    #20 0x00007ffff5db117b in _gtk_marshal_BOOLEAN__BOXED
    (closure=0x555556774390, return_value=0x7fffffffa6c0, n_param_values=<optimized out>, param_values=0x7fffffffa720, invocation_hint=<optimized out>, marshal_data=<optimized out>)
    at ../../../../gtk/gtkmarshalers.c:83
    #21 0x00007ffff550dc7d in g_closure_invoke (closure=0x555556774390, return_value=0x7fffffffa6c0, n_param_values=2, param_values=0x7fffffffa720, invocation_hint=0x7fffffffa6a0)
    at ../../../gobject/gclosure.c:810
    #22 0x00007ffff5521345 in signal_emit_unlocked_R
    (node=<optimized out>, detail=detail@entry=0, instance=instance@entry=0x555556961120, emission_return=emission_return@entry=0x7fffffffa840, instance_and_params=instance_and_params@entry=0x7fffffffa720) at ../../../gobject/gsignal.c:3635
    #23 0x00007ffff552994b in g_signal_emit_valist (instance=<optimized out>, signal_id=<optimized out>, detail=<optimized out>, var_args=var_args@entry=0x7fffffffa8f0)
    at ../../../gobject/gsignal.c:3401
[... cut ...]
#40 0x00007ffff79a18b0 in wxAppConsoleBase::OnRun() () at /usr/local/lib/libwx_baseu-3.1.so.2
#41 0x00007ffff7a3dbd5 in wxEntry(int&, wchar_t**) () at /usr/local/lib/libwx_baseu-3.1.so.2
#42 0x000055555579dfc7 in Slic3r::CLI::run(int, char**) (this=0x7fffffffde80, argc=<optimized out>, argv=0x7fffffffe088) at /home/ydelia/downloads/src/Slic3r/src/slic3r.cpp:471
#43 0x00005555557838fc in main(int, char**) (argc=<optimized out>, argv=<optimized out>) at /usr/include/c++/8/ext/new_allocator.h:79
````

Note that this was actually run with GDK_SYNCHRONIZE=1.

wavexx, 2019-03-17:
(verified on both wx 3 and wx 3.1)

bubnikv, 2019-03-19:
Would you please test with a5b202a ?

It looks like the gtk / OpenGL does not like activating the OpenGL context if the window is not visible. Weird.

Thanks.

wavexx, 2019-03-19:
Fixed, thanks!

closed 2019-03-19 by wavexx