prusa3d/PrusaSlicer #1958: Slic3r Calculations

touradnik9, 2019-03-16:
Hello,

I am using Slic3r Prusa Edition. Does anyone know the math behind slic3r's calculations when you export the g code?

Specifically the used filament in (m) and (mm^3).

Thanks!

[image]

bubnikv, 2019-03-17:
https://manual.slic3r.org/advanced/flow-math
https://en.wikipedia.org/wiki/Cylinder#Volume

closed 2019-03-17 by bubnikv

Sebastianv650, 2019-03-17:
Funny coincidence, I did some related investigations in the last days. Namely I measured the surface of test pieces along the printers Z direction with a Perthometer (a quite expensive professional device for measuring surface roughness).
Beside surface roughness data like Ra and Rz this also gives me a plot of the measured distance, quite simmilar as if you create a cross section of a single walled print and look at it through a microscope but with much better resolution.

To make a long story short, one of the findings was a confirmation what I had seen through my microscope when taking some pictures for the inconsitent walls thing some month ago: The edges of a line path when printed is not a half-circle as assumed by Slic3r and most likely any other slicer out there. Instead it's much flatter. @bubnikv if you are interested in the data to maybe improve Slic3r based on it I would be pleased to share them with you.
I'm also quite sure I found the main reason for "inconsistent walls" - not only on the MK3 as the hype was back then but on all 3D printers I have ever worked on. But that's off topic, if you are interested and want to invest time, feel free to come back and ask.

bubnikv, 2019-03-17:
> if you are interested in the data to maybe improve Slic3r based on it I
would be pleased to share them with you.

I am certainly interested. My time I can invest into this topic is limited
though.

> I'm also quite sure I found the main reason for "inconsistent walls"

I am certainly interested.

touradnik9, 2019-03-17:
Hello,

Thanks for providing the links, but my problem is that the numbers provided by slicer do not match according to any equation. For example, the picture I initially sent was created from an stl of length 100mm, height of 0.20mm, and extrusion thickness of 0.8mm. In no way am I able to calculate 28.54 mm^3. That is my issue.

The slic3r flow math page does not provide a way to determine the Radius of the half 'circles' formed on either side of the extrusion.

My guess is that it is more complex than one would think. For example, the surface roughness data you provided is probably a function of material type, viscosity, temperature, volumetric flow rate, nozzle diameter, layer height, speed, etc.

Sebastianv650, 2019-03-17:
You number doesn't fit at all. Maybe provide the gcode so we can see the values or even the 3mf file? I guess it's doing a full perimeter loop instead of a thin wall depending on your settings.

touradnik9, 2019-03-18:
Here is the gcode, but this is for a different file, as the one I initially posted about I did not save.

This one is showing 15.13 this as opposed to a value of 0.16

[image]

[GCODE.docx](https://github.com/prusa3d/Slic3r/files/2977439/GCODE.docx)

Sebastianv650, 2019-03-18:
For a single 100mm line of 0.2mm height and 0.8mm width the 15.13mm^3 are just as expected. Can't see an error. What did you expect?

touradnik9, 2019-03-18:
100 * 0.2 * 0.8 = 16 mm^3.
The reported value of 15.13 mm^3 is low, exceeding 5% in error.

I would think that is an exceptionally high error.  I am just wondering how does slic3r EXACTLY come up with 15.13.

Thanks
Tim

Sebastianv650, 2019-03-18:
Your extruded isn't extruding a square cross section, see the documentation links above. 

The formula is 0.2×(0.8-0.2)×L+(0.2/2)^2×PI×L where L is the distance.

touradnik9, 2019-03-18:
okay, thank you!!

This makes sense. Awesome!