prusa3d/PrusaSlicer #1959: Bridging angle startpoint and local bridges support

Kolous, 2019-03-16:
### Version
1.42a7

### Operating system type + version
Win7

### 3D printer brand / version + firmware version (if known)
i3 MK2.5 MMU2 (Single profil)

### Behavior

When trying to slice to valve cover (from ericpoolboy's Toyota engine) i ran into some strange bringing behavior. The default bridges (0°) are generated under some (30°). When i increase the value 90° is already correct and 180° gets me what I wanted the 0°. So the start value is wrong:

[image]

In the second step I tried to support the bridges in the middle by support entforcer:

[image]

Which supported the whole plane (was equal to support without bridge detecting) instead of generating only a slim bridge supporting pilar in the middle. if this is desirable behavior than a local bridge support can be considered as a feature request.

[image]

#### Project File (.3MF) where problem occurs

[VALVE_COVER.zip](https://github.com/prusa3d/Slic3r/files/2973582/VALVE_COVER.zip)

KnatteAnka, 2019-03-17:
quick question did you set the print setting to no support and then force support box?

Kolous, 2019-03-18:
You see in the first picture. Support only on build plate + don't support bridges + force support box.

FidelCapo, 2019-03-18:
I can reproduce issue with bridging angle.

About the supports. Your support enforcer is too low so it it does not intersect with object. When I move support enforcer up and set in support settings Auto generated supports and Don't support bridges to disabled I believe it is sliced as you want.

FidelCapo, 2019-03-18:
Created SPE-873

Kolous, 2019-03-18:
> About the supports. Your support enforcer is too low so it does not intersect with object. When I move support enforcer up and set in support settings - Auto generated supports and Don't support bridges to disabled I believe it is sliced as you want.

Sorry for the example file I did it multiple time to verify the behaviour. If you enlarge the enforcer it behaves the same. Let me rephrase it: the idea is to have bridges ("Don't support bridges" on not to support the whole bridge) but to generate only small support block for example in the middle to shorten the distance being bridged. The support enforcer should enforce so hard that it overrules "Don't support bridges" on area only defined by the shape of the enforcer (not whole plane).

FidelCapo, 2019-03-19:
I hope I understand correctly what you want from Slic3r. I attach modifed 3mf of your project. When you enabled Don't support bridges, Slic3r will not generate support under bridges even if the enforcer is in this place. Are supports in 3mf file like you want to generate them?

[VALVE_COVER-modified.3mf.zip](https://github.com/prusa3d/Slic3r/files/2982018/VALVE_COVER-modified.3mf.zip)

bubnikv, 2019-03-19:
In a nutshell: There is currently no way to support just part of the
bridge. You either have the whole bridge supported, or none of it (or just
the end points). This is how it currently works, and there is certainly
room for improvement, though there are around 600 other issues open right
now.

Kolous, 2019-03-19:
> 
> In a nutshell: There is currently no way to support just part of the bridge. You either have the whole bridge supported, or none of it (or just the end points). This is how it currently works, and there is certainly room for improvement, though there are around 600 other issues open right now.
> 

You have this input and your call is to accept it or not...

closed 2026-09-29 by prusaslicer-bot[bot]