prusa3d/PrusaSlicer #1961: Crash when clicking on "Slice now" button

guillaume-rico, 2019-03-16:
### Version
Version: 1.42.0-alpha7+win64
Build: Slic3rPE-1.42.0-alpha7+win64-full-201903011809

### Operating system type + version
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 17134), 64-bit edition

### 3D printer brand / version + firmware version
Stock MK3 - FW : 3.5.2-1999

### Behavior
Open Slic3R, import STL file in attachment (tuile_eau_dolphin.stl) then click on "Slice Now".
Slic3r crash without any message.

#### Project File (.3MF) where problem occurs

[tuile_eau_dolphin.zip](https://github.com/prusa3d/Slic3r/files/2973915/tuile_eau_dolphin.zip)

[tuile_eau_dolphin.3mf.zip](https://github.com/prusa3d/Slic3r/files/2973917/tuile_eau_dolphin.3mf.zip)

guillaume-rico, 2019-03-16:
This is a regression from version : 
Version: 1.41.3+win64
Build: Slic3rPE-1.41.3+win64-full-201902111723
Operating System: MSWin32
System Architecture: MSWin32-x64-multi-thread
Windows Version: Microsoft Windows [version 10.0.17134.648]

bubnikv, 2019-03-17:
yeah, I see it crashing in 1.42.0-alpha7.

bubnikv, 2019-03-17:
It is a known issue, which we shall fix. The dolphin stands on a thin tail fin, which is too thin to be printable, and Slic3r does not handle the empty layers just above the print bed.

Your model will slice if you enable "detect thin walls", but then it will likely not be physically printable, even with supports enabled.

bubnikv, 2019-03-17:
This is a similar issue to https://github.com/prusa3d/Slic3r/issues/1356
https://github.com/prusa3d/Slic3r/issues/1479

lukasmatena, 2019-08-26:
An extra check was added into PrusaSlicer2.1.0, so unprintable scenarios like this are detected before the application crashes. Closing.

closed 2019-08-26 by lukasmatena