prusa3d/PrusaSlicer #1963: Layer slider repaints heavily on mousemove events

wavexx, 2019-03-16:
### Version
Current master (954d571ab0a713d3df5fdcf97a9a9cc0e06fa471)

### Operating system type + version
Linux / Debian unstable (x11)

### Behavior
Slice something, then switch to the layer preview.
Change the current layer using the slider.

Now simply move the mouse over the slider widget: it repaints in full at every MouseMove event with noticeable flicker (in fact, you have to wait for the flicker to finish before you can use the widget).

I suspected an issue with wxWindows 3.0, but now I rebuilt 3.1.2 for testing and it has the same behavior.

YuSanka, 2019-05-15:
Thanks for your report, but unfortunately we can't reproduce your bug or probably can't understand the problem. Could you please retest your issue with last release and send us short video with your problem in case a nothing is changing.

vojtechkral, 2019-05-17:
I think this is the same problem as discussed later on in #1250 and hopefully should be fixed / improved by e354a306e68fad5e7e30f15b6c895a461bc06d1a .

wavexx, 2019-05-17:
@vojtechkral I couldn't test this change, in the last couple of days I get a GL error during startup:

````
[2019-05-17 18:52:56.658008] [0x00007fcad97756c0] [error]   OpenGL error in /home/ydelia/projects/Slic3r/src/slic3r/GUI/Camera.cpp:125, function apply_ortho_projection() : 1281 - Invalid Value
prusa-slicer: /home/ydelia/projects/Slic3r/src/slic3r/GUI/3DScene.cpp:52: void glAssertRecentCallImpl(const char*, unsigned int, const char*): Assertion `false' failed.
````

Should I open a new report for it?

wavexx, 2019-05-18:
Attaching video on the current master: https://www.thregr.org/~wavexx/tmp/VID_20190518_155524.mp4

@YuSanka You might have a compositor active that effectively hides the underlying issue?

bubnikv, 2019-05-20:
I just checked with two of our developers, who develop on Linux, and they never saw this behavior.

Do you get this flickering with our official binary builds?

What wxWidgets do you compile against? What GTK version? Is there anything special about your X server setup? Don't you run on Wayland?

wavexx, 2019-05-20:
On your binary build the effect is not visible. This is on plain X11. I've tested this behavior with wx 3.0 (debian unstable package) and wx 3.1 (rebuilt from git). I always defaulted to GTK3.

bubnikv, 2019-05-20:
Maybe it is an issue with wxWidgets / GTK3. Would you please try compiling
against GTK2? Thank you.

wavexx, 2019-05-20:
Indeed, wx 3.1 + GTK2 doesn't have this.

bubnikv, 2019-05-20:
cool, thanks for testing.

wavexx, 2019-05-20:
On Mon, May 20 2019, bubnikv wrote:
> cool, thanks for testing.

Are you interested in bug reports on wx+GTK3?

I don't mind reporting issues of this sort if you have eventually the
intention to move to GTK3.

Otherwise I can stick to wx+GTK2.

bubnikv, 2019-05-20:
We will eventually have to support GTK3, so we are interested. I shall just tag them with a new GTK3 tag.

wavexx, 2022-03-12:
Just wanted to add an update on this: I recently had to build PrusaSlicer on GTK3 due to multiple issues in HIDPI setups with GTK2 (setting labels do not respect the Xft.dpi settings - making them unreadable). Overall the GTK3 build works fine and solves font scaling as well.

This seems the only minor glitch I'm seeing so far.

One gotcha exists only during build of recent wxWidgets releases (https://github.com/supermerill/SuperSlicer/issues/1093#issuecomment-1046004099).

lukasmatena, 2022-03-14:
@wavexx Thanks, but I'm not exactly sure I understand. Do you still see the layer slider repainting issue with PrusaSlicer 2.4.1 or not? If you do, does it happen with our AppImages (that you can download from github) or only with your custom build?

This issue is 3 years old and GTK3 should now be fully supported.

wavexx, 2022-03-14:
@lukasmatena Yes, it does still happen.

- Add one object
- Slice (vertical slider appears)
- Moving on top of the slider is still fine (doesn't repaint)
- Click somewhere to move the vertical slider
- Now the slider repaints heavily even even on move events only

If you're running on linux, to make it visible you need to be running _without_ a compositor as that will doublebuffer-away the problem.

I tried the appimage a while back, but AFAIK that was building with GTK2 which didn't have this problem.

I'm also contributing to wxWidgets occasionally, so I'm running from upstream's sources. I already pushed some fixes for PS last year, so if there's any issue remaining I'd like to understand what's missing and fix it instead of building a custom copy.

I'm currently building with wx 3.1.5, with eglcanvas disabled (as mentioned above) and GTK3.
I don't think the repaint is due to upstream. It looks like over-aggressive clearing+redraw to me (maybe GTK2 was just faster), but being honest I didn't really look into why it does this.

lukasmatena, 2022-03-14:
@wavexx There is a GTK3 AppImage: https://github.com/prusa3d/PrusaSlicer/releases/download/version_2.4.1/PrusaSlicer-2.4.1+linux-x64-GTK3-202203101056.AppImage Can you please try that?

wavexx, 2022-03-14:
@lukasmatena so this doesn't seem to be repainting, but in order to be helpful to pinpoint the issue, where I can see the exact GTK3/wx sources being used for this build?

wavexx, 2022-03-14:
Ok, after a bit more searching through the build cmake, I've found the branch hopefully used for this build https://github.com/wxWidgets/wxWidgets/compare/master...prusa3d:v3.1.4-patched 

I don't see anything significant from a cursory look, but I'll try to build with these patches first and report back.

lukasmatena, 2022-03-14:
@wavexx Yes, we build against the branch v3.1.4-patched, which is 3.1.4+some things backported from upstream master+some more improvements, mainly dark-mode related.

If you want to play with it, I wish you a pleasant experience. We will probably not look into it if the static build against our patched wxWidgets works fine. Supporting all imaginable combinations on Linux is PITA and we would not do anything else.

wavexx, 2022-03-14:
@lukasmatena that's a rather bleak view ;). I don't see any pinned version of GTK. I assume this is coming from the image being used for the build. Can I also ask maybe where is the script which is used to build the appimage to begin with so I can extract all the relevant versions?

tamasmeszaros, 2022-03-14:
Hi @wavexx we are currently building the appimage in a Centos7 container. The exact version of gtk-devel is: 3.22.30