prusa3d/PrusaSlicer #1964: Support Modulus operator

dreamcryer, 2019-03-16:
### Version
1.42.0-alpha7+win64

### Operating system type + version
Win10 Pro 1809

### 3D printer brand / version + firmware version (if known)
Creality Ender 3
Klipper

### Behavior
When using modulus "%" operator in conditional g-code, g-code exporting results in error like this:
!!!!! Failed to process the custom G-code template layer_gcode
Parsing error at line 1: Not a boolean expression
{if layer_z % 10 == 0}M109 S220{endif}
    ^
!!!!! End of an error report for the custom G-code template layer_gcode

or 

!!!!! Failed to process the custom G-code template layer_gcode
Parsing error at line 1. Expecting tag literal-char
{if (layer_z % 10) == 0}M109 S220{endif}
             ^
!!!!! End of an error report for the custom G-code template layer_gcode

It seems that the parser does not support the modulus operator.
The modulus operator is very useful for generating test towers. Although we could work around this by using if-else statements with pre-calculated values, I feel it's more convenient to use the modulus operator.

Is this a new feature request? 
Not sure if the modulus operator is intentionally unsupported :)

bubnikv, 2019-03-17:
the parser is homebrewed, and only the minimum set of operators were
implemented, thus the modulo operator is missing. It should not be terribly
difficult to implement though.

smurfix, 2019-12-08:
It is now implemented. ;-)
Would you kindly merge my pull request (#3271)?

smurfix, 2020-01-21:
It has been merged. 👍

lukasmatena, 2020-01-22:
The request should indeed be fulfilled after #3271 was merged. Closing.

closed 2020-01-22 by lukasmatena