prusa3d/PrusaSlicer #1965: Feature Request: Different extruder for external perimeter and top/bottom layers

Virindi-AC, 2019-03-17:
It would be nice to be able to print all of the parts of the model which are visible from the outside with one extruder, but still have internal perimeters and [perhaps only some] solid layers printed with something else. The reason for this is that when using stronger/stiffer materials such as carbon fiber fills, your parts are limited in appearance (they must be black or gray).

If you could print only the external perimeter (or two) and some top/bottom layers with colored filament, but have other perims and infill carbon fiber, you could have nearly the strength of a carbon fiber part but with any desired appearance.

Adrian-at-CrimsonAuzre, 2019-05-22:
I would love this option, I am looking into building a dual hot end printer and equipping it with something like a .4mm nozzle and a .8 mm nozzle. If I could use the .8mm nozzle for infill/internal perimeters and the .4 for external perimeters and fine details I would be ecstatic.

Asrothear, 2019-05-31:
Same here ^^ Would be a great Option

mihai-stanimir, 2021-02-25:
This would be really useful.

closed 2026-09-29 by prusaslicer-bot[bot]