prusa3d/PrusaSlicer #1966: Feature Request: Split into parts option for cut

Kolous, 2019-03-17:
### Version
1.42a7

### Operating system type + version
Ubuntu 18.04

### 3D printer brand / version + firmware version (if known)
i3MK2.5 MMU2

### Behavior
When performing cut both parts are dropped imediately to the bed:

[image]

This is in most cases desired behaviour when you split object to be glued later. But in this specific case I wanted to divide object in two parts for MMU automatic filament change (instead of manual colorchange). The cutted part can't be mover on Z axes and always drops after move to the bed.

Feature requests:

1) Cut into parts option - perform cut and generate two parts for MM printing with keeping their position
2) Allow to move one cuted part on top of other (when split into two objects)
3) Add function of automatic colorchange for MMU (different extruders to be used for different layers)

bubnikv, 2019-03-17:
You are better off using the modifier meshes and assigning an extruder ID to the modifier mesh.

KnatteAnka, 2019-03-17:
Or just import the STLSs in the first STL part file and set extruder there.
or if already one STL open the part and hit the button splitt part now you can set color and ZXY is not affected.

Kolous, 2019-03-18:
> 
> 
> You are better off using the modifier meshes and assigning an extruder ID to the modifier mesh.

Splitting one object by plane into two components seems to me much more straitforward.

Kolous, 2019-03-18:
> 
> 
> Or just import the STLSs in the first STL part file and set extruder there.
> or if already one STL open the part and hit the button splitt part now you can set color and ZXY is not affected.

Not sure what are you refering to. I'm talking about modification of one object in Slic3r not importing multiple components.

Premik, 2019-07-27:
I'm also for having a possibility to disable the auto-arrangement on cut. It comes useful sometimes. Perhaps it could be disabled by default when only one part (upper or bottom) is being kept.

Today I wanted to print 8 copies of the same object. The handle of the scissors should be red.

- I added material change somewhere in the middle to get the top layer of the handle in red:
[image]

- Then I cut the parts leaving only the top-most layer.

[image]

Printed the first print on top of the second one leaving the first print on the bed from the previous print. Because I naturally assumed the cut wont rearrange the things I ended up with lots of these:

[image]

Not exactly what I wanted :)

bubnikv, 2019-08-01:
The upcoming PrusaSlicer 2.1.0-alpha1 will allow you to assign extruder for a Z range. That should cover this request at least partially.

@Premik I am not quite sure I understand. Please provide a 3MF file so we can test your scenario.

Premik, 2019-08-03:
What I did is not so common use-case. Basically objects on the first picture I posted appear to be on the same positions as on the second picture. But actually on the second picture they are slightly closer together. I would prefer them not to change their positions when using the cut-tool.

ssvenn, 2019-08-18:
It looks like the new "edit layers" function in 2.1 will solve this issue nicely, but would it be possible to add a visual way to adjust the height like the orange cube on the Cut gizmo? It would make the process faster and more intuitive I think.

[image]

bubnikv, 2019-08-19:
> would it be possible to add a visual way to adjust the height like the orange cube on the Cut gizmo? It would make the process faster and more intuitive I think.

@enricoturri1966 ?

bubnikv, 2021-06-03:
Automatic MMU color change has been implemented long time ago at the slider of the G-code preview in a similar way to a non-MMU color print.

closed 2021-06-03 by bubnikv