prusa3d/PrusaSlicer #1967: Z Offset with MMU

marlinian, 2019-03-17:
Slic3rPE 1.41.3

Windows 10 

Prusa i3 Mk3 MMU2

When importing models into the slicer, it would be really useful if I could apply a Z offset and also to be able to assign a specific filament to separate models on the build plate.  The Z offset is in the settings but not able to be changed.  

_Is this a new feature request?_Yes

#### Project File (.3MF) where problem occurs
_Upload a Slic3r PE Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r PE 1.42.0-alpha and newer)_

bubnikv, 2019-03-17:
> When importing models into the slicer, it would be really useful if I
could apply a Z offset and also to be able to assign a specific filament to
separate models on the build plate.

The Z-offset in Slic3r is there to compensate for the bed leveling.
Instead of Z offset on objects you likely want to cut the object with a
plane.
Also there IS a setting to set the extruder per object:
in 1.41.3 you need to double click on the object
in 1.42.0-alpha7 (or the upcoming beta) you are able to change the extruder
per obect in the object list.

bubnikv, 2019-03-24:
Does it work for you?

bubnikv, 2019-05-04:
no response, closing

closed 2019-05-04 by bubnikv