prusa3d/PrusaSlicer #1968: [no Issue] Findings regarding Slic3r math and "inconsistent extrusion"

Sebastianv650, 2019-03-17:
This is the measurement data I announced in #1958. As this is a longer one, I think it should have it's own issue number.
I'm doing such investigations from time to time, as I often don't find answers to my questions. Usualy around 3D printer development I miss (available) scientific research, best example was the so called inconsistent extrusion thing, where tons of messages where created by users but the usable content was low to state it in an optimistic way.
But keep your blood pressure low, my tests where not done due to this topic, in fact I wanted to get roughness values for aerodynamic flow calculations around 3D printed wings. But somethimes one things connects to another one.

**Test Setup:**
I printed single and double perimeter hollow test cubes on multiple 3D printers:
- Prusa MK2S
- Lulzbot TAZ 5
- German RepRap X400

I used normal 1,7 and 3mm filament and also Prusament. So the results are not aiming specific to Prusa, but are valid in general.
The results where consistent across all printers, also the closer-tolerance Prusament made no visible changes. But to remove variables, all shown numbers and graphs in the following content are from my MK2s using Prusament.
Measurements are done with professional device for surface roughness measurement, a so called Perthometer. It works like an old gramophone on steroids: A very thin needle is pulled along the test piece and the vertical deviation is measured in µm (1/1000mm). In fact, it is calibrated to give valid results to 0.1µm or 1/10000mm!

**Result 1: Shape of extrusion-line edges**

The graphs confirm what I also have seen under a microscope some month before: The edges of a perimeter line, looking at the cross section of a print, are not shaped like a half circle with `R = Layer height / 2` as at least Slic3r but most likely all slicers are assuming.
Istead, with a layer height of 0.2mm the average radius is about 0,122mm (122% of the expected value). With a layer height of 0.1mm the average radius is about 0,066mm (132% of the expected value)
The measurement is done by pixel measurement from the scanned plots.

This information might be useful to create better calculations for the amount of extruded plastic and also the line spacing needed.
Here is the plot of the 0.2mm layer height example used. Note that the x and y axis have different scales:
[image]
and the 0.1mm one:
[image]

Here is a section of the 0.2mm one where I scaled the two axes 1:1 for better visualisation:
[image]

**Result 2: "Inconsistent extrusion"**

As you might have already noticed in the two graphs above, there is a clear pattern visible in the 0.1mm print: There is a high spot on every ~7 layer which is also visible in the print when angled to the light correctly. This pattern is not visible on high layer heights like 0,2mm or 0,25mm, but gets more and more prominent at 0.15mm and below.
As the test parts are orientated along X and Y, we can ignore these axis as the source for the inconsistency. They are positioned in the same direction to the same position on each layer, therefore if there would be a problem it's highly unlikely to see such a continous pattern.

There are 3 effects I can think of to create this:
- Z wobble
- Inconsistent extrusion itself
- Inconsistent Z positioning

First I wanted to know if this effect is caused by a wobble, therefore the perimeters would have the same, correct width but are placed inaccurate above each other or if the placement is perfect and the width of the lines itself varies.
Therefore I did a measurement of a 1 perimeter test piece, measuring the exact same position but once on the outer side of the wall and once on the inside. To eleminate possible microstepping inaccuracies I choosed a layer height of 0.08mm which is a multiple of full steps on the Z axis. You might know this as magic numbers. This magic numbers wasn't changing the result at all by the way, which is good new for microstepping.

As you can see, the inconsitency is due to line width changes and not due to a wobble. I flipped the lower chart upside-down so it's equal to a real look at a printed 1 perimeter wall from the side:
[image]

What can we note due to that information:
- Z wobble can't be the problem. First the 7*0.1mm layer pattern we see isn't matching up with the thread pitch, second the issue is not a wobble but a width change.
- Extruder can't be the problem. As with the Z wobble, we should see a change in extrusion width with each half revolution of the extruder when the hobbed part would be out of axis. As I calculated, the amount of plastic extruded during the pattern doesn't match up with the circumference of the hobbed bolt. I calculated the amount of axial deviation needed to get a width change as seen in the plots and compared that to the real runout of my MK2S extruder. It doesn't fit by the order of a magnitde!
- **Inconsistent Z positioning** is what looks like the best fit. By measurements of the plots it's clear that there is some height variant. With the given values, it seems that the z height can only be reached +-3µm, therefore a layer height of 0.2mm varies between 0.197mm and 0.203mm. This is a realy tiny amount, but all it needs to create the variable extrusion widths. And this is also the reason why we only see the artefacts in prints with low layer heights, 3µm on 0.2mm layer height is nearly nothing. 3µm on a 100µm print has doubled it's influence.

This chart is showing the effect based on the layer height, which fits quite good with the plots:
[image]

I hope this information can be used for further investigation on real printers. I don't have the tools for measuring 3µm of z deviation or influence it. What I have tried is to lower the Z speed and acceleration to insane low level, without any change in the results. Maybe a finer leadscrew pitch could help.

Notifying @bubnikv.

bubnikv, 2019-03-17:
Thanks a lot, it will take time to digest.

mikolaszuza, 2019-03-17:
Very interesting read, thank you!

wavexx, 2019-03-25:
Very interesting. I assume there's no Z hop going on here. Still, I find the repeating perplexing. Any chance you could provide the data in digital form and with a taller wall? (0.8 layers, 32mm tall?).

Sebastianv650, 2019-03-25:
No z hop was active. There is no digital output from the device, I can only plot out the graphs. There are always only a few mm of measurement, I think the graphs above where from 5.6mm distances but only 4mm are shown in the graph.

As I have only a 0.4mm nozzle, I can't do 0.8mm layer heights, sorry.

wavexx, 2019-03-25:
On Mon, Mar 25 2019, Sebastianv650 wrote:
> No z hop was active. There is no digital output from the device, I can
> only plot out the graphs. There are always only a few mm of
> measurement, I think the graphs above where from 5.6mm distances but
> only 4mm are shown in the graph.

Mmmh...

You said you already looked through a microscope. What sort of setup did
you use? Did you manage to see a cross-section (if so, how?).

I ask because I work in a microbiology lab, and I tried a few times to
look at printed samples (notably, at nylon/cf to see the fiber
orientation), but I cannot think of a convenient way to see the same
effect.

> As I have only a 0.4mm nozzle, I can't do 0.8mm layer heights, sorry.

Meant 0.08 here as you did previously, sorry.

Sebastianv650, 2019-03-26:
For the microscope, I also printed single perimeter cubes. Then I cut a stripe out of one wall section and sanded the cutting edge to real 90° as good as possible. Finaly I "painted" the edge with a black permanent marker for better contrast.
In the end I had a section of the wall about 10mm height and maybe 2 or 3mm wide. I placed this piece on my microscope so that when looking from above I see the 10mm by 0.5mm cut section. This was actualy the hardest part, it took some tries with tweezers to get it in the right place without it falling to the side.

wavexx, 2019-03-27:
On Tue, Mar 26 2019, Sebastianv650 wrote:
> For the microscope, I also printed single perimeter cubes. Then I cut
> a stripe out of one wall section and sanded the cutting edge to real
> 90° as good as possible.

Using PLA?

I tried doing this on a few occasions, but the sending process in
whatever-grit always ruined the face.

This might work with ABS or ASA by cutting using a razor blade instead.
But the MK3 currently sits enclosed in my office and I really don't want
to print that stuff in here.

> permanent marker for better contrast. In the end I had a section of
> the wall about 10mm height and maybe 2 or 3mm wide. I placed this
> piece on my microscope so that when looking from above I see the [...]

That's tricky, but if I understand the process you were just looking at
the side face from above.

In this case, I could try to print a 1 wall rounded cylinder instead and
lay it on a side so that edge artifacts are removed and making it stand
is no longer an issue.

All else being equal, I should still be able to see the repeating
pattern without having to cut or stand.

bubnikv, 2019-03-27:
> With the given values, it seems that the z height can only be reached +-3µm
> This chart is showing the effect based on the layer height, which fits quite good with the plots ...

That may very well be the explanation of the issue 602, thanks for all your time to investigate it and document it to us.

Some claim that the stepper motor is not precise if micro stepped at very high micro stepping. IMHO the issue with the torque of the stepper motors is the following: There is no torque produced by the stepper motor, if the rotor is at the target position. The motor only produces some torque in a slip, the higher slip the higher is the torque. Now if the motor is just slightly before / behind the target angular position, the torque is very low, and it may not be sufficient to overcome the drag of the mechanical system.

Now the torque depends on the angular position in relation to the phase between the two magnet coils, the highest torque is achieved if both coils are energized equally. Therefore the stepping accuracy should be highest at full steps, if both coils are equally energized at these full steps. Just entering the "magic number" will not guarantee that condition. I have no idea whether stepping at the full steps only will show any positive effect. Also the MK2 / MK3 firmware moves the Z axis to compensate for an uneven print bed, which also may have some negative effect in theory, though we have tried to print with Z compensation disabled and the effect was the same.

supermerill, 2019-03-27:
> though we have tried to print with Z compensation disabled and the effect was the same.

but was  z=0 at a full step?

bubnikv, 2019-03-27:
I don't remember. I remember switching the Einsy board into a 16 micro
stepping mode though.

wavexx, 2019-03-27:
On Wed, Mar 27 2019, bubnikv wrote:
> Some claim that the stepper motor is not precise if micro stepped at
> very high micro stepping. IMHO the issue with the torque of the
> stepper motors is the following: There is no torque produced by [...]

This should be mitigated by Z hop though.

To me the interesting fact is the underlying effect that generates the
harmonic as shown. It doesn't seem to be directly related to the lead
screw/steps.

Sebastianv650, 2019-03-27:
@wavexx yes I used PLA. Maybe I also finaly used the unsanded ones, I can't tell for shure to be honest.

> In this case, I could try to print a 1 wall rounded cylinder instead and lay it on a side so that edge artifacts are removed and making it stand is no longer an issue.

You mean like a section of a barrel? But then you are looking at a curved object, so it's not as easy to see the deviations.

@bubnikv indeed it would be an interesting test to see if something changes with disabled bed compensation and microstepping disabled completely for Z. Then using a propper value like 0.16 or 0.08mm as a layer height. Is it needs changing the FW config I don't want to do it now, but maybe I will try it in some days.
At a thread pitch of 8mm, 3µm equals 0.135° angle deviation. Also very hard to measure directly, but might be possible. And even if friction and the angle deviation from stepper due to the needed torque is the problem, I can't think of an easy way to fix it at the moment.

wavexx, 2019-03-27:
@Sebastianv650 not exaclty a barrel:

[image]

the curved edges are there just to prevent ringing and corner artifacts

bubnikv, 2019-03-28:
> At a thread pitch of 8mm, 3µm equals 0.135° angle deviation. Also very
hard to measure directly, but might be possible.

A nice indicator is to make two flags by gluing two pieces of glue tape to
the trapezoidal screws.

supermerill, 2019-03-28:
Dirty trick to measure very small rotation:

building:
1- Print a 10mm-radius disk with a hole for the lead screw. Make it stick really well on your lead screw.
2- Create a support for your top-of-the line high dpi laser mouse, to make it stay over the side of the disk.
3- create (or find) a quick program that just register the mouse move every ms (or more if needed), write it into the standard output and reset the position to 0,0. (I can do that if you request it, it's really quick to do).

measuring:
move z by 1 mm, get the result and add all numbers to see by how many pixel 1mm of z move. 
execute your gcode test file, convert pixels to mm & plot the test data.

some inputs:
16000 dpi => 629pixel/mm,
For a disk of 10mm radius, 0.1° ~= 0.1745 mm => 109.78 pixels
We can have a precision of 0.001° with a little 20mm disk. I don't know the margin of error, though.

wavexx, 2019-04-03:
Had some busy days, but I managed to do a test print of the solid above and take a picture under my stereoscope, mostly for testing if the approach works. This is 50x, 0.2 layers and light from the side.

[image]

The solid is sitting on top of a black sheet to enhance the contrast. I think I should be able to get a more shallow solid (~3mm) that would allow me to go up to 500x, which should be plenty to replicate the above, but using visual sampling. I only fear dust might be an issue for this approach. I wasn't careful with this one, but just blowing air doesn't really remove any of it.

closed 2026-09-29 by prusaslicer-bot[bot]