prusa3d/PrusaSlicer #1969: Change pattern for overhangs

espr14, 2019-03-17:
### Version
1.42.0 alpha 7

### Operating system type + version
Windows 7

### 3D printer brand / version + firmware version (if known)
i3 MK3

### Behavior
* Overhang loops cannot be printed well (attached picture).
* Slice the attached STL to reproduce the issue.
* Slicer should avoid loops or circles and print straight lines where possible. With straight lines you can print all shapes with convex holes without any loops.

[image]
[pad.zip](https://github.com/prusa3d/Slic3r/files/2975151/pad.zip)

_Is this a new feature request?_
YES

### Details
No supports were used. It could be printed right even without supports. Layer height 0.25.

Slicer uses the same angle for the first two (overhang) layers which makes print even worse. Third layer is in 90° angle. So it looks like it starts changing the angles from the second layer, not the first one.

Shapes with convex holes can be printed with straight lines only using bridges. Shapes with non-convex holes can be printed by adding bridges next to each other. I'll try to create illustration.

supermerill, 2019-03-18:
If you look at **https://github.com/supermerill/Slic3r/wiki/No-perimeters-on-bridge-area**,  is one of the four options the one you describe? Or can you write a drawing?

espr14, 2019-03-18:
No. I meant this:

[image]

As you can see there is no curve, just straight lines. All lines are supposed to be in a single layer. Slight overlapping is not necessary but can improve stitching of lines. Crucial thing is cooling of long lines (bridges). If they are not cooled enough, they would stay bent and crossing lines won't stick to them.

espr14, 2019-03-18:
And this is solution for non-convex shapes. Overlap at the start and the end of move improves stitching. Wipe out and the end also cuts the string so it does not travel with the tip to another place.

[image]

espr14, 2019-03-26:
This is how Prusa designs and prints his parts. In the first layer it prints bridges in one direction, in the second layer bridges in another (perpendicular) direction and in the third layer it prints the final shape. This is a way how to overcome the issue.

[image]

Solution in slicer would include these settings:
1. (Minimal) layer height (or taken from printer/filament settings). That's because we want the first imperfect layers to be as thin as possible. It kind of irremovable support.
2. Number of layers of bridges before printing final shape (with possible (short) overhangs).

Solution should analyze all layers before printing final shape to minimize overhangs when printing the final shape. Then it should sort bridges from the longest one to the shortest. The longest bridge should be printed in the last layer (before the final shape is printed). Typically, the longest bridge will be shortened by previous layers which is an intention. That's because we want to minimize overhangs as well as length of the bridges.

espr14, 2019-04-02:
Another example of overhang hole which won't be printed well.

[image]

espr14, 2019-04-11:
Printing of alternating bridges would also help with printing of domes. They have the same issue - full overhang layer with a hole in mid-air.

bubnikv, 2019-04-13:
We have had this discussion before somewhere at exactly this issue tracker. What you are suggesting is Slic3r to modify the input geometry. This is not only complex, but also it may lead to some results that the user does not expect.

bubnikv, 2019-04-13:
Your proposal is similar to https://github.com/prusa3d/Slic3r/issues/1860 in the sense that it wants Slic3r to modify the model.

espr14, 2019-04-13:
Yes, all these linked issues try to solve the same issue. There are 3 approaches:
1. Modify the model so it prints more and more in each succeeding layer. In all but last layer use bridges only.
2. Modify the model like in 1. but print the place it in minimal layer height. In case of 0.2 print it can print 4 layers of 0.05 in one.
3. Don't modify the model but print it in different way (see above).

I suggest all three possibilities. The 3rd approach also solves non-convex geometry. The #1860 and #127 propose 1st approach only.

bubnikv, 2019-04-13:
Would you please provide the implementation?

espr14, 2019-04-13:
That's easy. Suppose we have library which can:
1. For defined walls and space to be filled a procedure can find all non-crossing bridges in optimal angle. That means it fills as most space to be filled as possible. Similar procedure is already in the Slic3r.
2. Find the longest line which sticks to already built output and fill the space which should be filled. Both ends of the line has to be stitchable to already built part. See the 3rd image in this issue thread.

Algorithm is the same for all the 3 approaches, just for the 1st and 2nd number of steps will be limited and adding of bridges will be done just once per layer and lines will be added if no bridges were found. This is the version for the 3rd approach:

```
while (outputHasChanged){
    addOptimalBridges()
}
while (outputHasChanged){
    addOptimalLine()
}
```

If possible print another part of model between bridges' sets and lines to make them cool enough.

bubnikv, 2019-04-13:
Then please provide a pull request.

espr14, 2019-04-13:
Lot of code to get familiar with.

espr14, 2019-04-18:
I've created a special model so the Slic3r does what I propose. The goal is to print bridge with a hole. Normally, it would create a lot of overhangs with sharp corners. This tweaked model prints no corners in the air but 2. Obviously, it could be done even without any corner in the air (but not by tweaking).

Traditional approach:
[image]

Proposed approach:
[image]

[overhang hole.3mf.zip](https://github.com/prusa3d/Slic3r/files/3096280/overhang.hole.3mf.zip)

closed 2026-09-29 by prusaslicer-bot[bot]