prusa3d/PrusaSlicer #1970: Don't support short bridges

espr14, 2019-03-17:
Filament setting should include definition how long is a short bridge. While short bridges need not to be supported, long bridges should be. Supports should be generated for long bridges only.

After adding option to filament menu, you can remove "Don't support bridges" in support menu.

jonnieZG, 2019-07-13:
Ditto!

There should be an option "Do not support bridges shorter than _selectable_value_"

Inarisee, 2020-07-26:
@rtyr Seeing this issue being open for so long - leaves me frustrated... How can we make Prusa considering this request? For me and also the author of #3859 this is a major issue.

bubnikv, 2021-02-22:
Closing this one, keeping Bridge max linear length must be capped in "don't support bridges" #3859 open

closed 2021-02-22 by bubnikv