prusa3d/PrusaSlicer #1971: After building on linux I get an "assert" error

herrwusel, 2019-03-17:
### Version
```bash
$ git describe --tag
version_1.42.0-beta-3-gf5b30237e
```

### Operating system type + version
Kubuntu 18.10

### Behavior
From time to time I update Slic3r and so far everything went well. So this time I removed the build directory and created a new one. 
```bash
$ cmake .. -DSLIC3R_FHS=1
$ make -j6
$ sudo make install
``` 
finished normal. When I open Slic3r I get this and it forgets about previous settings:
```
ASSERT INFO:
../src/generic/treectlg.cpp(2205): assert ""item"" failed in SelectItem(): SelectItem(): invalid tree item

BACKTRACE:
[1] wxGenericTreeCtrl::SelectItem(wxTreeItemId const&, bool)
[2] Slic3r::GUI::Tab::update_page_tree_visibility()
[3] Slic3r::GUI::Tab::update_visibility()
[4] Slic3r::GUI::Tab::load_current_preset()
[5] Slic3r::GUI::GUI_App::load_current_presets()
[6] Slic3r::GUI::GUI_App::OnInit()
[7] wxEntry(int&, wchar_t**)
[8] Slic3r::CLI::run(int, char**)
[9] main
[10] __libc_start_main
[11] _start
```
I can press to continue and Slic3r will launch but without any previous settings.

herrwusel, 2019-03-18:
Found the solution: the home directory changed from .Slic3rPE-alpha/ to .Slic3rPE-beta/. I copied all files to the new location and had to do the initial set up. After this the error dissappeared. 

Maybe it would be better to have one config directory for slic3r?

closed 2019-03-18 by herrwusel

bubnikv, 2019-03-19:
>   Maybe it would be better to have one config directory for slic3r?

That will certainly happen after release candidate.