prusa3d/PrusaSlicer #1976: 1.4.2 Beta_OSX clicking travel resets layer selection in preview

andrewbecker, 2019-03-17:
### Version
1.42.0-beta+

### Operating system type + version
OSX 10.14.3

### 3D printer brand / version + firmware version (if known)
NA

### Behavior
Slice a model and go to preview.  Slide the layer slider down to a layer towards the middle of the print.  Click the checkbox next to travel to select or deselect it.

* _Expected Results_
The travel lines should be shown or hidden, but the layer slider should not reset.
* _Actual Results_
The layer slider is reset back to the top layer.  This behavior is not seen when selecting retractions, unretractions, or shells.
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
_Upload a Slic3r PE Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r PE 1.42.0-alpha and newer)_

enricoturri1966, 2019-03-18:
Working on it -> http://dev.prusa3d.com/browse/SPE-875

bubnikv, 2019-03-24:
it will be fixed in the beta1.

closed 2019-03-24 by bubnikv