prusa3d/PrusaSlicer #1980: Slic3r 1.42.0 beta x64 crashing on slice after resizing a object

marteinnsig, 2019-03-18:
### Version
Slic3r 1.42.0 beta x64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa MK3 Firmware 3.7.0-RC1

### Behavior
After importing a object, and doing a rezise that is not locked to all axes (eg, only resize on Z ) for example, the program chrashes as I hit "slice" button

_Is this a new feature request?_
No
[box-test.zip](https://github.com/prusa3d/Slic3r/files/2979137/box-test.zip)

#### Project File (.3MF) where problem occurs
_Upload a Slic3r PE Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r PE 1.42.0-alpha and newer)_

bubnikv, 2019-03-19:
I believe it is the same issue as #1978 #1979
There is a bug newly introduced into the beta, where there is a division by
zero happening for tiny objects.

bubnikv, 2019-03-24:
it will be fixed in beta1. closing

closed 2019-03-24 by bubnikv