prusa3d/PrusaSlicer #1985: 1.42 beta crash when deleting printer profile created by configuration assistant.

ctilley83, 2019-03-18:
### Version
1.42-beta+

### Operating system type + version
Mac OS X 10.14.3

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
If you create a custom printer profile using the configuration assistant, then try to delete that profile, it will cause a crash. It does not crash if you create a printer profile manually then delete it.  Bug only occurs when creating a custom profile with the configuration assistant.

1. Use the configuration assistant and select the option to create a user profile
2. Go to the Printer Settings tab and select that newly created printer profile
3. Click delete. (this will cause a crash)

upon reopen the printer profile is successfully deleted. Possibly trying to reference memory that is already been released back to the OS.

FidelCapo, 2019-03-19:
Thank you for report. I can reproduce this issue.
Created JIRA-877

bubnikv, 2019-03-24:
Deleting of printer profiles was quite buggy. It did not take into account dependencies of the profiles (print / filament profile being dependent on the printer profile), and it did not take into account the printer technology either. It shall be fixed in the upcoming beta1. closing.

closed 2019-03-24 by bubnikv