prusa3d/PrusaSlicer #1986: UI Color scheme should accommodate color blind

GWdd, 2019-03-18:
### Version
Slic3r_1.42.0-alpha7+win64

### Operating system type + version
Windows 7 x64 

### 3D printer brand / version + firmware version (if known)
Prusa Mk3

### Behavior
http://www.colourblindawareness.org/colour-blindness/

Colors used to denote user ability levels are difficult for color deficient persons.  Yellow and lime green for example are indistinguishable for many forms of red-green color blindness.

Color blindness affects 12% of the male Caucasian population, and about 8% in a substantial way. Red-green is most common (1 in 12).  Females and other populations are affected less so. 

https://ghr.nlm.nih.gov/condition/color-vision-deficiency#statistics

When designing the user interface, choose colors that are substantially different and more saturated rather than relatively close shades or pastel tones.  

Common problematic color sets are yellow and green; green and orange; blue and purple.

In rainbow cases, like color choices for temperature, it is difficult to find a good solution. But in other areas, like the Simple/Advanced/Expert selection, the Yellow / Green choices are needlessly close in color: fine for color normals, but hard to use for us color deficient persons.  For me,  yellow / orange / red would make more sense.  Worth noting: green / orange /red would be as bad as yellow/green/red.  

I can't tell anyone what I see; except to say that I can't see what you see;  And 1 out of 12 men can't see what you see, either.  

All I can do is ask that you don't make a product more difficult for others.  Be aware of the design issue, and make choices better suited for the full population of intended users.

In this image: there are a lot of yellow and green marks - I couldn't tell you - with certainty - which are which.

[image]

bubnikv, 2019-03-19:
Actually we have a color blind person in our testing team :-) and likely he first found out in his adult time for the first time when he was looking at exactly this screen. And luckily for you, we have a solution already. Please let us know whether this is sufficient to you.

[image]

[image]

GWdd, 2019-03-19:
The two colors are more or less indistinguishable to me unless I concentrate - and only indicative if they are both present - even then I'd need a key; to my vision, one is a darker shade of the other.  That orange green thing.  Or maybe they are both green, or both orange, I can't really tell.  A deeper orange red would help. 

Normally I prefer leaf green to lime green; but when orange is nearby, even leaf green isn't helpful.  I'm in that class of people who don't see reds very well.  Went months not having the red gun on my monitor and only noticed when I had some free time and discovered I was only getting black queens in Black-Jack.  I'm most likely in this class (protanop):

[image]

bubnikv, 2019-03-19:
You can click on these buttons and change the color.

GWdd, 2019-03-19:
Ah - didn't recognize they were buttons.

closed 2019-03-19 by GWdd