prusa3d/PrusaSlicer #1988: Print infill lines in better order

espr14, 2019-03-19:
Bottom/top infill layer is separated into continuous lines. Order of these lines cannot be specified and is somehow random. The problem is continuous cooling of material. Adjacent lines should be printed one after the other. If not, lines do not fuse into each other which causes weaker result and worse optical quality.

Lines should be ordered according to time of print. It is not possible to do the best always so it should be some best effort algorithm. However, following example is not printed in the best order (the small gap is filled at the end of the layer):

[image]

espr14, 2019-03-20:
Whole place is in a single layer but printing caused these optically uneven surfaces.

[image]

guestisp, 2019-03-23:
I saw the same. I think even cura does in this "strange" way

bubnikv, 2019-03-24:
There is simply no continuous line that could fill this region (at least with the rectilinear infill pattern).

guestisp, 2019-03-24:
I'm using gyroid and does the same

espr14, 2019-03-24:
It cannot be filled by a single line but it can be filled by 2 lines. Here it looks like it is filled by 4 lines!!!

espr14, 2019-03-27:
> I'm using gyroid and does the same
Shell infill is not

bubnikv, 2019-05-04:
Duplicate of https://github.com/prusa3d/Slic3r/issues/471

closed 2019-05-04 by bubnikv