prusa3d/PrusaSlicer #1989: Crash on cut+separate

wavexx, 2019-03-19:
### Version
Current master (78ac357d3438347ccd3833b4603940e2ab6b01a5)

### Operating system type + version
Linux (debian unstable, X11)

### Behavior
While investigating #1955 I accidentally run into another crash. This one reads more similar to #1464, but it's underlying cause is a double-free inside libtess, not due to a BadMatch (which has been fixed).

This is triggered by:

- slicing an object
- one of the resulting parts can be furthermore split into sub-objects
- attempting to split into parts results into actual crash

#### Project File (.3MF) where problem occurs
[test.zip](https://github.com/prusa3d/Slic3r/files/2984766/test.zip)

- Load "mk3-lm8-bracket.stl"
- Perform a cut using the default position
- The second resulting part has two separated legs: select it
- Attempt either split into parts or split into objects: crash due to double-free

There's nothing special about this model, except that it allows a split-to-parts after the cut. I reproduced the same on several other larger models. This is the smallest I could get my hands on.

However, I quickly whipped up a test case in openscad (test.stl), where I have an "U" object where I could do the same, but it doesn't reproduce ;)

valgrind refuses to run on slic3r for me right now, but I still suspect the test object could be useful.

port513, 2019-03-19:
It also crashes when trying to slice a one layer model

wavexx, 2019-03-19:
On Tue, Mar 19 2019, Lord Ulf Henrik Holmberg wrote:
> It also crashes when trying to slice a one layer model

It would be helpful if you could create a separate report with a sample
one-layer model on it.

lukasmatena, 2019-03-20:
Thanks for the report. I managed to reproduce the issue. I get a crash in Slic3r::TriangleMesh::split() on line `if (! facet_visited[facet_idx])`, because facet_idx is -1 for some reason. It's quite possibly a result of some bug/cornercase in the cut() function. We will certainly look at it when we find time. (I have created http://dev.prusa3d.com/browse/SPE-878 and assigned it to myself.)

What makes you think the crash is inside libtess? The tesselation library is called during cutting, so I assume the crash would happen sooner, without giving you the opportunity to split. I compiled with GCC address sanitizer which revealed no other issues before that crash (I'm on Linux, though).

wavexx, 2019-03-20:
On Wed, Mar 20 2019, lukasmatena wrote:
> What makes you think the crash is inside libtess? The tesselation
> library is called during cutting, so I assume the crash would happen
> sooner, without giving you the opportunity to split. I compiled with
> GCC address sanitizer which revealed no other issues before that crash
> (I'm on Linux, though).

I was using an optimized build, so maybe something was wrong or the
stack trace was incomplete [?]

Just out of curiosity, barely related (and since you're running on linux
as well): can you manage to run slic3r through valgrind/memcheck at all?
When I tried yesterday valgrind failed on me (vex temporary storage
exhausted). Never had this happen before.

lukasmatena, 2019-03-20:
I tried to run valgrind on Slic3r just for you and it works, but spits out huge loads of errors similar to
```
==6671== Invalid write of size 4
==6671==    at 0x17FA85A7: ??? (in /usr/lib/x86_64-linux-gnu/dri/i965_dri.so)
==6671==    by 0x17FAB361: ??? (in /usr/lib/x86_64-linux-gnu/dri/i965_dri.so)
==6671==    by 0x17D10948: ??? (in /usr/lib/x86_64-linux-gnu/dri/i965_dri.so)
==6671==    by 0x17D01C0A: ??? (in /usr/lib/x86_64-linux-gnu/dri/i965_dri.so)
==6671==    by 0x17AC35D7: ??? (in /usr/lib/x86_64-linux-gnu/dri/i965_dri.so)
==6671==    by 0x17AADC3B: ??? (in /usr/lib/x86_64-linux-gnu/dri/i965_dri.so)
==6671==    by 0x17ABFEF2: ??? (in /usr/lib/x86_64-linux-gnu/dri/i965_dri.so)
==6671==    by 0x179ACE67: ??? (in /usr/lib/x86_64-linux-gnu/dri/i965_dri.so)
==6671==    by 0x2F5A120: Slic3r::GUI::GLCanvas3D::_render_background() const (GLCanvas3D.cpp:4575)
==6671==    by 0x2F4CFF6: Slic3r::GUI::GLCanvas3D::render() (GLCanvas3D.cpp:2478)
==6671==    by 0x2F555BD: Slic3r::GUI::GLCanvas3D::on_paint(wxPaintEvent&) (GLCanvas3D.cpp:3818)
==6671==    by 0x2F905FE: wxEventFunctorMethod<wxEventTypeTag<wxPaintEvent>, Slic3r::GUI::GLCanvas3D, wxPaintEvent, Slic3r::GUI::GLCanvas3D>::operator()(wxEvtHandler*, wxEvent&) (event.h:373)
==6671==  Address 0x7f1e355b2504 is not stack'd, malloc'd or (recently) free'd
```
I did not and will not investigate it further. If you're using gcc/clang (I assume you are), I strongly recommend using the ASAN tool. Just run cmake with `-DSLIC3R_ASAN=1` parameter and you are good to go (you can use Release build). I'm not an expert to compare it to valgrind, but this is what we use.

wavexx, 2019-03-20:
On Wed, Mar 20 2019, lukasmatena wrote:
> ==6671==    by 0x2F5A120: Slic3r::GUI::GLCanvas3D::_render_background() const (GLCanvas3D.cpp:4575)
> ==6671==    by 0x2F4CFF6: Slic3r::GUI::GLCanvas3D::render() (GLCanvas3D.cpp:2478)
> ==6671==    by [...]

Yeah, these are quite normal. At least on debian there's a default
suppression list which is quite comprehensive and excludes most of the
lower level libraries.

Thing is, valgrind doesn't even get past the startup failing with an
internal error for me, which is odd. But anyway.. ASAN is also fine too.

SijmenSchoon, 2019-03-25:
I am having the same problem trying to split [this model](https://github.com/tltx/MeC64/blob/master/keycap_adapter/Cherry_to_commodore_print_v1_1_binary.stl). 

According to comments in the code that generates the `neighbors`, `-1` means that it's unconnected and so I assume it does not need to be split. Changing the for loop at TriangleMesh.cpp:440 to 

```cpp
for (int j = 0; j < 3; ++ j) {
    auto neighbor = this->stl.neighbors_start[facet_idx].neighbor[j];
    if (neighbor >= 0)
        facet_queue.push(neighbor);
}
```

seems to fix the issue - splitting and cutting both work now.

lukasmatena, 2019-03-27:
@SijmenSchoon It is indeed just this check that was missing. The fix can be (almost) copy-pasted from the function `number_of_patches` that is just above, so the solution is literally in front of us (sadly, this code is repeated several times).

It is a small fix but if you want to make a PR, you can do. Just add the second condition too and please don't use the `auto` keyword in a context simple as this.

SijmenSchoon, 2019-03-27:
@lukasmatena guess I can see if I clean up the duplicate code while I'm at it

PR #2042 (2019-04-01): Fix crash on splitting some models
There was a bunch of duplicate and dead code related to splitting models, so I cleaned up the duplicate code and removed the dead code.

I also fixed some warnings in the related files while I was at it, so errors would not get drowned in the build output.

Fixes #1989
closed 2019-04-01 by bubnikv