prusa3d/PrusaSlicer #1990: Delete part/modifier

espr14, 2019-03-19:
### Version
1.42.0 alpha 7

### Operating system type + version
Windows 7

### Behavior
I can add a part or a modifier to an existing object but I don't know how to delete it. Delete key does nothing. Also there is no delete item in context menu of the part/modifier (right click).

Could you add possibility to delete part/modifier/support enforcer/blocker?

lukasmatena, 2019-03-20:
It should be possible to delete a modifier mesh by selecting it (either in scene or in the right panel) and then clicking delete icon in the top menu or pressing delete key. The only catch is that the delete key only works if the scene has focus (i.e. clicking in the object list disables the key until the mouse returns into the scene, at least briefly). This is a known issue caused by technical reasons related to passing the keyboard event properly. We will investigate how to solve this issue.

Please, let us know if that was the problem or if there is any other bug.

bubnikv, 2019-03-20:
> It should be possible to delete a modifier mesh by selecting it (either
in scene or in the right panel) and then clicking delete icon in the top
menu or pressing delete key.

Both work on Windows 10, and both shall work on all the other systems
supported.

espr14, 2019-03-20:
Selecting object in the 3D view and pressing delete key or icon works. But it does not work in this case:
1. unselect all (click to empty space in 3D view)
2. select modifier object in the list
3. icon is greyed and cannot be clicked, also delete key does nothing

You have to select the modifier in the 3D view, or select something else in the list and then select the modifier and press delete icon.

You could add "Delete part" next to "Change part" in the context menu of the part/modifier so there would be 3 ways how to delete it.

bubnikv, 2019-03-24:
Delete now works in the object list as well, with the exception of OSX, where the Delete is now accessible from the context menu. This change will be part of the beta1.

closed 2019-03-24 by bubnikv