prusa3d/PrusaSlicer #1992: New modifier creator

espr14, 2019-03-19:
Existing modifiers can be box, cylinder or sphere. However, it could be great to have another way. User would select starting layer, end layer and a point. This point works as a seed which determines selected part.

Let's have pyramids on the common base. With this modifier you can easily pick single pyramid (without common base) and change infill, perimeters or other parameters. It is not dependent on the shape so it is easy also for hexagonal pyramids or cogwheels. With current shapes it would be much harder or even impossible. Since it does not cut object vertically, more settings can be changed than in case of standard shapes.

lukasmatena, 2023-11-12:
This basically describes the Layer Range Modifier, implemented years ago. Closing.

closed 2023-11-12 by lukasmatena

espr14, 2023-11-12:
Layer Range Modifier takes complete plane in XY but this takes only single object in that Z range.