prusa3d/PrusaSlicer #1993: Unnecessary support material

Kolous, 2019-03-20:
### Version
1.42b

### Operating system type + version
Win7 64bit

### 3D printer brand / version + firmware version (if known)
MK2.5 MMU2

### Behavior
Unnecessary support material put on top of the support interface layers and outside of object.

[image]

#### Project File (.3MF) where problem occurs
[Bell_Housing.zip](https://github.com/prusa3d/Slic3r/files/2986633/Bell_Housing.zip)

espr14, 2019-03-20:
The support material in the holes is OK. If you lower pattern spacing to 0.1 it will do what you think, but it will increase time and consumption. It is caused by discretization of the space to squares. If a small part of a square is used for support then whole square is built. Therefore, the hole is filled.

The only solution is to change the algorithm from squared support to an organic one. This is more complicated and computing will take much more time.

The supports on the sides are definitely an algorithm error.

Kolous, 2019-03-21:
And the one more layer on the top of the interface layer (shown by arrow)?

espr14, 2019-03-21:
Looks like it is by purpose. Some kind of fence but don't know what is the purpose. However, it is so small so it does not hurt. Maybe it is used for easier support removal.

Kolous, 2019-03-22:
And one two more issues with supports. Default profile settings used.
- interface layers in the air (red)
- too small support to be printed/usefull (blue)

[image]

bubnikv, 2021-03-22:
Testing snug supports

[image]

bubnikv, 2021-03-23:
Fixed with af9c7c967f08fcf22ffa628c324c90eda65230f1

    Implementing a new switch for the shape of support towers:
    expanded to a grid (the old way) vs.
    snug (like the upstream Slic3r, Cura or Ideamaker).
    
    Snug supports suffered from the degeneracies when merging overhang islands
    over a large number of layers when projecting the support towers down.
    We borrowed the idea & a bit of code from Cura by simplifying the support
    polygons by closing the concave cracks, see the smooth_outward() function
    and the MutablePolygon class.

closed 2021-03-23 by bubnikv