prusa3d/PrusaSlicer #1994: machine_max_acceleration_retracting ignored

guestisp, 2019-03-20:
Version: 1.42.0-beta+linux64 but probably even older one.

I've set `machine_max_acceleration_retracting` as 1500mm/s and `machine_max_acceleration_retracting` as 500mm/s but the generated gcode only add a single `M204 P500 R1500 T500 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2`

There is a little bit confusion here. The tooltip says that `machine_max_acceleration_retracting` is `M204 T` but `M204 T` is the travel acceleration, the retract acceleration is `M204 R`

So, how can I set a travel acceleration different from the retract acceleration ?
Slic3r seems to wrongly use the same value for both, regardless the tooltip.

I think a field is missing:

`machine_max_acceleration_extruding` is correct as `M204 P`
`machine_max_acceleration_retracting` tooltip is wrong. The proper gcode command is `M204 R` but `M204 T` is shown in tooltip
`machine_max_acceleration_traveling` is missing. The proper gcode command is `M204 T` but currently is exported with the same value as `machine_max_acceleration_extruding`

http://marlinfw.org/docs/gcode/M204.html

guestisp, 2019-03-20:
Trying a simple fix right now. If it build, i'll post a PR

bubnikv, 2019-03-24:
See https://github.com/prusa3d/Slic3r/issues/1089 for the discussion

guestisp, 2019-03-26:
Thank you. But tooltip is still wrong.

bubnikv, 2021-07-29:
Marlin's M204 PRT implemented in PrusaSlicer 2.4.0-alpha1 by splitting the Marlin firmware flavor to "legacy" and "Marlin 2"

closed 2021-07-29 by bubnikv