prusa3d/PrusaSlicer #1996: 1.42.0-alpha7: extruder is reset on "reload from disk"

Mirarkitty, 2019-03-20:
### Version
1.42a7

### Operating system type + version
Linux

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
* import stl
* set extruder to something else than default
* reload stl from disk
* extruder is reset to "default"

* expected extruder to stay at the same setting
* (resulting in printing with the wrong extruder and/or having to redo work to set extruders)

* especially annoying on multi-extruder objects
[image]
[image]

bubnikv, 2019-03-24:
The feature does not work correctly indeed
https://github.com/prusa3d/Slic3r/issues/552
https://github.com/prusa3d/Slic3r/issues/1471

caro-coder, 2019-10-31:
This is very annoying! So many times that I had to cancel a print, because I noticed that I again forgot to set the right extruder after reloading from disk ...

Mirarkitty, 2019-11-03:
I must say doing "reload from disk" on an object with several source files (multi extruder print) doesn't quite give anywhere near the result you expected, either.

lukasmatena, 2019-11-04:
@Mirarkitty If you're able to build PrusaSlicer yourself, try the latest master build. @enricoturri1966 has recently improved the reload from disk feature a lot.

bubnikv, 2019-12-18:
The "Reload from disk" feature is reworked in the upcoming PrusaSlicer 2.2. Closing.

closed 2019-12-18 by bubnikv

Mirarkitty, 2020-01-11:
Seems to work now. Thanks