prusa3d/PrusaSlicer #1997: No support interface layers generated

Kolous, 2019-03-20:
### Version
1.42b

### Operating system type + version
Win7 64bit

### 3D printer brand / version + firmware version (if known)
MK2.5 MMU2

### Behavior
When using support entforcer no support interface layers are generated.

[image]

[image]

#### Project File (.3MF) where problem occurs
[Transmission_housing.zip](https://github.com/prusa3d/Slic3r/files/2989434/Transmission_housing.zip)

bubnikv, 2019-03-24:
Are not the dense support layers missing below bridges? Do you have the "don't support bridges" option enabled?

Kolous, 2019-03-24:
"don't support bridges" option is not influencing this behaviour. I think this issue is related to support enforecer. But not only: [here is observed similar issue](https://github.com/prusa3d/Slic3r/issues/2006)

bubnikv, 2021-03-17:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/2801

bubnikv, 2021-03-18:
I have reworked the support generator, so the support enforcers take precedence over "supports on build plate only" and over "dont' support bridges". I have also extended the support generator to trim the support towers by the object silhouette so that for the automatic supports and "supports on build plate only" enabled, the support towers will not land on top of an object, with the drawback of possibly being too tiny / flimsy to be printable.

[image]

bubnikv, 2021-03-18:
[image]

bubnikv, 2021-03-19:
Fixed with 9f09f03228478aa77549efec845717b06e46cbc3

closed 2021-03-19 by bubnikv

ProTubeVR, 2021-03-22:
> 
> 
> Fixed with [9f09f03](https://github.com/prusa3d/PrusaSlicer/commit/9f09f03228478aa77549efec845717b06e46cbc3)

Just builded your last change to test on my side,
Nothing changed (the support structure look updated, but not generating interface layers below 70° overhang, in fact it's worst
[image]
to me problem isn't solved at all, or your changes haven't been merged?

bubnikv, 2021-03-22:
@ProTubeVR As commented elsewhere, the dense support interface is only created below low angled surfaces. The other surfaces are supported sufficiently with the sparse support.

Your model is different from the original issue, thus it is a different issue. If you believe this is a problem, then please open a new issue and attach a 3MF so we can test. Thank you.