prusa3d/PrusaSlicer #1998: Importing .AMF file invokes "Detected object outside print volume" error/warning

Jebtrix, 2019-03-20:
### Version
Version 1.42.0-beta-win64

### Operating system type + version
Windows 7 PRO 64bit

### Behavior
Importing .AMF objects now displays a "Detected object outside print volume. Resolve a clash to continue sliding/export process correctly" at bottom of 3D Editor View. This problem is new to 1.42 beta, older versions import the same .AMF file centered as intended.

Once you import the .AMF file and get the error message, the object is not displayed or selectable anywhere at any zoom level. I have to click [Arrange] to bring object into the 3D Editor View.

After further investigating the root cause of the now problematic .AMF files is it includes a <constellation> xml element; ie:
` <constellation id="0">
        <instance objectid="47304"/>
 </constellation>`

[5x5x5mm.Test.Cube_not_working.AMF.txt](https://github.com/prusa3d/Slic3r/files/2990036/5x5x5mm.Test.Cube_not_working.AMF.txt)

If I strip out the constellation element the object imports as intended.

[5x5x5mm.Test.Cube_modified_working.AMF.txt](https://github.com/prusa3d/Slic3r/files/2990045/5x5x5mm.Test.Cube_modified_working.AMF.txt)

enricoturri1966, 2019-03-25:
Working on it -> http://dev.prusa3d.com/browse/SPE-883

FidelCapo, 2019-03-27:
Thank you for report. This issue should be fixed in next release.
Closing

closed 2019-03-27 by FidelCapo