prusa3d/PrusaSlicer #1999: Export STL

bootdsc, 2019-03-21:
### Version
_Version of Slic3r Prusa Edition used goes here_
1.42.0-beta+linux64
_Use `About->About Slic3r` for release versions_

### Operating system type + version
Linux mint 19 xfce

### Behavior
Export STL only exports the model without the generated supports or base plate. Shouldn't everything be exported, supports included? If i want to preserve the model with a certain set of supports but not as a sliced file then the option to export with supports seems logical.

_Is this a new feature request?_
This is either a error or a missing feature.

neophyl, 2019-03-22:
the stl is only the model.  Sounds like you need to save the project as a 3mf if you want to save the whole environment to work on later.

bubnikv, 2019-03-22:
Do I understand it correctly, that you want to export the STL with the SLA pad and tree supports?
Why do you need that?

verner33ob, 2019-03-27:
I don't know, why he needs that, but i know, why i would need that. Maybe his reasons are similar.

I have a very complex hollow model.
If i uncheck "support on build plate only", slic3r generates tons of inside support, that i don't want and that i also can't delete.
If check "support on build plate only", slic3r doesn't generate supports between an extended feater (like an arm) and the build plate.
So i would like to create the supports for my unhollowed object, export the stl, delete my unhollowed object in a 3d-program, then then import my hollowed object and only the support back to slic3r.
That way i have ALL the needed supports on the outside and no unneeded supports on the inside.

neophyl, 2019-03-27:
Verner to do what you describe the normal method is to add your hollow model, enable supports, and then use support enforcer/blocker modifiers to control where slicer adds supports.  In this way you can 'delete'the supports by telling slicer not to generate them there in the first place.

You can export the project as a .3mf to save it for later with all the modifiers if at some point you want to edit things.

bootdsc, 2019-03-27:
Yes i would like to export the model with the supports and pad as a single stl. There's more then one scenario where this is useful. If i want to upload a model to share and it needs difficult to place supports then i prefer uploading a ready to slice model with supports already built. Another reason is to print the model on multiple brands of resin printers. Hardly anyone only has one printer and its typical to buy a different brand when upgrading. When i have trouble printing a model i send it to a friend and have them take a look, maybe help with support placement or try printing on their machine. Export with supports is a option with B9 Creator, nanoDLP, Chitubox, Formware and i imagine many others. If you want Slic3r pe to be on the same level as everyone else then having all of the standard features is a good idea.

verner33ob, 2019-03-27:
@neophyl 
i know that's a wonderful way to do it.
Unfortunately in SLA-Mode there's no option to add support enforcer/blocker

bootdsc, 2019-03-27:
And that's the issue.

Domi96, 2019-04-08:
Acutally it is very usefull too share complex prints with supports as STL. So everyone can slice them with the layer height they want, and STL files are by far not that big as multiple sets of png files with high resulution would be. Therefore I hope that this feature will be added in the feature.

tigerpolis, 2019-04-09:
Yes please add this!!!

X3msnake, 2019-04-09:
does exporting to 3mf keeps the supports? if so it is halfway there to the stl export. and yes STL export is a desirable and needed function.

bubnikv, 2019-04-10:
3MF keeps the custom support points.The tree structure is auto generated
from these points.

bootdsc, 2019-04-10:
So if you change the support settings or update to a new version of slic3r thats had changes related to supports its going to be a totally different structure. Exporting with the generated supports ensures its the same every time and allows to share the file.

bubnikv, 2019-05-04:
@enricoturri1966  is looking into this issue.

X3msnake, 2019-05-06:
@bunikv if the supports are generated on screen they can be exported as STL

enricoturri1966, 2019-05-06:
With https://github.com/prusa3d/Slic3r/commit/a3385278e5e892e644a02a288c1cf9c8bd296d00 a new command has been added to Export menu which allows to export generated SLA support and pads to STL

X3msnake, 2019-05-06:
Awsome Masters

bubnikv, 2019-05-22:
Implemented, closing.

closed 2019-05-22 by bubnikv

RHTizzy, 2019-07-12:
Option "export plate as stl including supports" is greyed out, even after generating g-code. This is not intuitive, what must be done to make use of this feature?

buzkill420, 2020-02-26:
> 
> 
> Option "export plate as stl including supports" is greyed out, even after generating g-code. This is not intuitive, what must be done to make use of this feature?

BUMPED.. please and thank you

bubnikv, 2020-02-26:
>  Option "export plate as stl including supports" is greyed out, even
after generating g-code. This is not intuitive, what must be done to make
use of this feature?

This feature is only active in SLA mode and when the SLA supports are
already generated.

dryphi, 2020-12-01:
Well that's silly. The whole point was to export the model with the TREE SUPPORTS which are only available in SLA mode.