prusa3d/PrusaSlicer #2000: Slic3r crashes when I try to slice

manddar, 2019-03-21:
### Version
N/A - I use generic 3D printer

Slic3rPE-1.42.0-beta+win64-full-201903171406 Release

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Anycubic i3 Mega with Marlin firmware

### Behavior
Slic3r crashes when I try to slice this stl.

https://www.thingiverse.com/thing:2134978/files (Circular)

_Is this a new feature request?_
No

Petclaud, 2019-03-21:
Same here. Tried to slice a bed level calibration design. Only one layer high. Once I click on the 'Preview' button, Slic3erPE crashes. No issues with the Alpha 7 release.

bubnikv, 2019-03-21:
it is an embarrassing tiny error introduced into the beta. It triggers on
small FDM objects only (SLA is fine).
We will provide an updated beta today or tomorrow.

F0x06, 2019-03-22:
Same on 40x40x40 cube in vase mode

gabtremblay, 2019-03-22:
Same problem here with Slic3rPE-1.42.0-beta+linux64-full-201903171406

`
(gdb) r
Starting program: /home/initnull/Applications/3DPrinting/Slic3rPE-1.42.0-beta+linux64-full-201903171406/bin/slic3r 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
[New Thread 0x7ffff36aa700 (LWP 13138)]
[New Thread 0x7ffff2ea9700 (LWP 13139)]
[New Thread 0x7ffff26a8700 (LWP 13140)]
[New Thread 0x7ffff1ea7700 (LWP 13141)]
[New Thread 0x7fffe3fff700 (LWP 13143)]
Multisample enabled
[Thread 0x7ffff26a8700 (LWP 13140) exited]
[New Thread 0x7ffff26a8700 (LWP 13144)]
[New Thread 0x7fffe2ffe700 (LWP 13145)]
[Thread 0x7fffe2ffe700 (LWP 13145) exited]
[Thread 0x7ffff26a8700 (LWP 13144) exited]
[New Thread 0x7fffe2ffe700 (LWP 13152)]
[New Thread 0x7fffe27fd700 (LWP 13153)]
[New Thread 0x7fffe190b700 (LWP 13154)]
[Thread 0x7fffe27fd700 (LWP 13153) exited]
[Thread 0x7fffe2ffe700 (LWP 13152) exited]
[Thread 0x7ffff1ea7700 (LWP 13141) exited]
[New Thread 0x7ffff1ea7700 (LWP 13155)]
[New Thread 0x7fffe27fd700 (LWP 13156)]
[New Thread 0x7fffe2ffe700 (LWP 13158)]
[New Thread 0x7fffe0f0a700 (LWP 13159)]
[Thread 0x7ffff1ea7700 (LWP 13155) exited]
[Thread 0x7fffe2ffe700 (LWP 13158) exited]
[Thread 0x7fffe27fd700 (LWP 13156) exited]
[Thread 0x7fffe190b700 (LWP 13154) exited]
[New Thread 0x7fffe27fd700 (LWP 13160)]
[New Thread 0x7fffe2ffe700 (LWP 13161)]
[New Thread 0x7fffe190b700 (LWP 13162)]
[New Thread 0x7ffff1ea7700 (LWP 13163)]

Thread 16 "slic3r" received signal SIGFPE, Arithmetic exception.
[Switching to Thread 0x7fffe27fd700 (LWP 13160)]
0x0000000000a9b5d1 in Slic3r::GCodeAnalyzer::_calc_gcode_preview_retractions(Slic3r::GCodePreviewData&, std::function<void ()>) ()
(gdb) bt
#0  0x0000000000a9b5d1 in Slic3r::GCodeAnalyzer::_calc_gcode_preview_retractions(Slic3r::GCodePreviewData&, std::function<void ()>) ()
#1  0x0000000000a9b917 in Slic3r::GCodeAnalyzer::calc_gcode_preview_data(Slic3r::GCodePreviewData&, std::function<void ()>) ()
#2  0x0000000000ac3ddd in Slic3r::GCode::do_export(Slic3r::Print*, char const*, Slic3r::GCodePreviewData*) ()
#3  0x0000000000955520 in Slic3r::Print::export_gcode(std::string const&, Slic3r::GCodePreviewData*) ()
#4  0x0000000000c42ed7 in Slic3r::BackgroundSlicingProcess::process_fff() ()
#5  0x0000000000c436d8 in Slic3r::BackgroundSlicingProcess::thread_proc() ()
#6  0x0000000000c43996 in Slic3r::BackgroundSlicingProcess::thread_proc_safe() ()
#7  0x00007ffff7eab424 in std::execute_native_thread_routine_compat(void*) (__p=<optimized out>) at /build/gcc/src/gcc/libstdc++-v3/src/c++11/thread.cc:94
#8  0x00007ffff66b1a9d in start_thread () at /usr/lib/libpthread.so.0
#9  0x00007ffff6149b23 in clone () at /usr/lib/libc.so.6
(gdb) 
`

guestisp, 2019-03-23:
The same here

bubnikv, 2019-03-24:
This should be fixed in beta1. Closing.

closed 2019-03-24 by bubnikv