prusa3d/PrusaSlicer #2001: Missing/disappearing preset profiles for print settings

haleypearse, 2019-03-21:
**Version**

slic3r PE 1.42.0-alpha5+
Mac OS 10.14.2 Mojave

**Behavior**

All of the default and custom presets for print settings have disappeared, regardless of preferences checkboxes such as "suppress default presets" or "show incompatible ... presets." This requires me to redo all print settings every time I open slic3r. The filament and printer settings presets are normal. If I save a preset, it will show in the drop down menu, but none others. I found the preset .ini files in the application support folder, but I can't get them into slic3r. 

[image]

bubnikv, 2019-03-22:
http://projects.ttlexceeded.com/3dprinting_slic3r_profiles.html
http://projects.ttlexceeded.com/3dprinting_slic3r_profiles.html#set-print-setting-dependencies

haleypearse, 2019-03-28:
> http://projects.ttlexceeded.com/3dprinting_slic3r_profiles.html http://projects.ttlexceeded.com/3dprinting_slic3r_profiles.html#set-print-setting-dependencies
> […](#)

Thanks, but this didn't solve the problem. I also updated to 1.42.0 beta1+full and again lost all presets, including filament and printer settings also.

bubnikv, 2019-03-28:
beta stores profiles into a Slic3rPE-beta directory.

haleypearse, 2019-04-03:
Thanks, indeed, copying profile .ini files into the beta directory made them show up in slic3r – but not for print settings! There, I still only see the last saved profile. 
[image]
[image]

bubnikv, 2019-04-03:
The Prusa system profiles are all contained in a single config bundle,
stored in the vendor directory. Also the "installed" printers are then
marked in slic3r.ini.
IMHO you should just run the wizard and install the missing system profiles.

haleypearse, 2019-04-04:
Thanks. I may try that. However, I'm currently able to save presets again, and have them continue to be available, which is much more important than the system profiles.

bubnikv, 2019-04-05:
would you please close this issue if it is solved for you? Thans.

haleypearse, 2019-04-05:
Oh, yes, but I closed it already 15 days ago, because it was in the wrong repo. Is there more I need to do?
[image]

bubnikv, 2019-04-05:
thanks, I overlooked that you closed the issue.

ChasonDeshotel, 2022-03-27:
>IMHO you should just run the wizard and install the missing system profiles.

That's not really a solution though... I mean... I'm running into the same issue, almost 3 years later on SuperSlicer (which is a fork of Prusa). And sadly that doesn't bring them back anyway

For posterity--I believe this is the bug case: https://github.com/prusa3d/PrusaSlicer/issues/8039. Filed 20 days ago and awaiting information. Just wanted to link these two cases in case anyone else ends up here from the googles