prusa3d/PrusaSlicer #2002: Feature request: different filament presets for color changes

beeb, 2019-03-22:
The color change feature in the 1.42 beta is amazing and makes it very easy and fast to add filament changes. However, this is thought for color changes while keeping the same material (and often brand) for the various parts.

I think this feature could really benefit from being able to choose different filament presets for each color change. That way, people could use different materials, temperatures, cooling settings for the various parts. This becomes extremely important when using different brands of filament (even within the same material type) because they often require different printing temperatures because of the grade being different. The diameter could also be different from one filament to the next which would require manual tuning of the flow parameter (and manual calculation of how diameter should affect the flow variable).

ftrofin, 2020-11-23:
Any progress on this? It's been more than an year since this issue was opened...

INSOMNIAC84, 2020-11-23:
I'd allso leave a vote for this request.
Even with different Filaments of the same vendor and material i usually end up with different Profiles that, at least have different retraction or flowrate parameters than the other. 

Getting a dropdown list after clicking the plus in layer view, where the user could select another Filament profile would be realy handy there. Maybe toggled by an "advanced" Checkbox to keep the current wellknown "simple" workflow as default.

Zettymaster, 2021-09-12:
I ran into this today, just to show there are more people that want this (if not obvious by the many other duplicate issues).

My specific use case is that I want a change from TPU to PLA and back to TPU, creating an anti slip surface on the top and bottom.

Using the Macro language is very cumbersome and without knowing GCode to begin with its even more unpractical.

Mikmikbt, 2022-01-14:
I think this could be an important feature. 
Many people maybe they do not know the real potential of this function. Personally I find it very useful and it seems absurd to have to use Simplify3d every time I need it. At the moment I believe that prusaslicer is quite complete but with this functionality it would be, I think, perfect. Hope others can join in with their own requests. Thank you

timur-tabi, 2022-06-09:
It's been three years since this simple feature was suggested and there's been no action on it.  I really could have used this feature, as I was printing with ASA and PETG on one project.  Fortunately, both materials work at the same temperature, so I was able to make my print work, but that won't always be true.

For the curious, my ASA is white and my PETG is black, and that's why I used them.

mahksy, 2022-06-30:
I agree with @timur-tabi on this, this would be incredibly useful for me aswell, sadly nothing changed since three years.

averydelmiller, 2022-12-13:
Sorry to bump this thread, but I think this would be amazing. I wondered why this hasn't been implemented yet, and I imagine there are a number of problems to consider. (If using a single extruder, heating and cooling the nozzle each layer will really extend printing times!) However, I think this would be really useful. 
I'm working on printing PETG supports for PLA prints as an alternative to break-way support materials.

gbeatty, 2023-01-09:
Same issue for me. I'd like to print NinjaTek TPU on top of ABS. The settings are wildly different between the two and being able to choose a material profile at a certain layer height would be amazing!

fuzzybat23, 2023-03-19:
LOL, I just made a post requesting this same thing. It's a good thing they have time to close requests saying their duplicates, but not implement a VERY GOOD IDEA that was requested over 3 years ago.

timur-tabi, 2023-03-19:
Dude, implementing new features is a lot more work than closing bug reports, you can't compare the two.

fuzzybat23, 2023-03-19:
I've done a little coding in my time. A feature like this would be simple considering 90% of the code is already written. Know what that code is? The color change. Adapting that to allow a filament change shouldn't take 3 years. So yes, I did make that comparison and I feel. It's quite accurate.

mahksy, 2023-04-27:
Well, maybe you wanna do a little coding in this time aswell, and implement a simple feature and open a PR?
Would be pretty simple, right? <:)

NVNDO, 2023-05-30:
1st suggestion open since **4 years** - In the meanwhile 15 duplicates and all closed

I assume the suggestion is still considered as it remains open, but instead of just silently closing all duplicates, giving an answer to weather this is actually considered and / or will be implemented in the near future would be more than helpful.

So could someone @rtyr please give as something... would be nice :)

#10694 was mine btw.

NVNDO, 2023-06-06:
I might actually have a walk-around for you guys as nobody seems to really care about our request. 

If you trigger M600 for a filament change, you can add whatever gcode before that in gcode settings.
To adjust for a different filament I use add **before M600** (all gcode after that is not triggered):

Speed:
M220 S<factor>

Purge %:
M221 S<factor>

Temps: 
M104 S<temperature>  M109 S<temperature>

Retraction: (not yet tested)
M207 S<retraction_distance> F<retraction_speed> Z<retraction_z_lift>

That's at least how I do it...
Hope this helps anyone

- Happy printing -

codefaux, 2023-11-21:
Is there some gcode placeholder method we can use to at least figure out which color change we're on?

closed 2026-09-29 by prusaslicer-bot[bot]