prusa3d/PrusaSlicer #2003: Assert error on startup

wavexx, 2019-03-22:
### Version
Current master (f5516f24c444d26c0c791f4c239eb35ba60f9f37) 

### Operating system type + version
Linux (debian unstable)

### Behavior
This was fixed earlier on this week (see #1971) but it reappeared.
Happens both with existing presets and with a fresh setup (just removed ~/.Slic3rPE-beta to confirm).

ASSERT INFO:
./src/generic/treectlg.cpp(2205): assert ""item"" failed in SelectItem(): SelectItem(): invalid tree item

BACKTRACE:
[1] wxGenericTreeCtrl::SelectItem(wxTreeItemId const&, bool)
[2] Slic3r::GUI::Tab::update_page_tree_visibility()
[3] Slic3r::GUI::Tab::update_visibility()
[4] Slic3r::GUI::Tab::load_current_preset()
[5] Slic3r::GUI::GUI_App::load_current_presets()
[6] Slic3r::GUI::GUI_App::OnInit()
[7] wxEntry(int&, wchar_t**)
[8] Slic3r::CLI::run(int, char**)
[9] main
[10] __libc_start_main
[11] _start

SijmenSchoon, 2019-03-22:
I seem to have found a fix, going to PR.

bubnikv, 2019-03-24:
@YuSanka would you please look into the pull request?

What wxWidgets are you compiling against? What GTK version?
I quickly browsed through your pull request, but frankly it looks as a workaround of some wxWidgets / GTK issue, not as a fix of some bug of ours.

ArchangeGabriel, 2019-03-25:
I have the almost the same one:
```
ASSERT INFO:
./src/generic/treectlg.cpp(2216): assert "item" failed in SelectItem(): SelectItem(): invalid tree item

BACKTRACE:
[1] wxGenericTreeCtrl::SelectItem(wxTreeItemId const&, bool)
[2] Slic3r::GUI::Tab::update_page_tree_visibility()
[3] Slic3r::GUI::Tab::update_visibility()
[4] Slic3r::GUI::Tab::load_current_preset()
[5] Slic3r::GUI::GUI_App::load_current_presets()
[6] Slic3r::GUI::GUI_App::OnInit()
[7] wxEntry(int&, wchar_t**)
[8] Slic3r::CLI::run(int, char**)
[9] main
[10] __libc_start_main
```

This is from a source-compiled 1.42-beta1, using wxgtk3 3.0.4 from ArchLinux.

ArchangeGabriel, 2019-03-25:
I’ve also seen https://github.com/prusa3d/Slic3r/issues/1971, but can’t be affected since I did not have that `.Slic3rPE-alpha` dir in the first place.

SijmenSchoon, 2019-03-25:
> I have the almost the same one:
>
> ```
> ASSERT INFO:
> ./src/generic/treectlg.cpp(2216): assert "item" failed in SelectItem(): SelectItem(): invalid tree item
>
> BACKTRACE:
> [1] [...]

I got this issue too after fixing the first one, so that's also included in my [PR](2005)

SijmenSchoon, 2019-03-25:
> @YuSanka would you please look into the pull request?
> 
> What wxWidgets are you compiling against? What GTK version?
> I quickly browsed through your pull request, but frankly it looks as a workaround of some wxWidgets / GTK issue, not as a fix of some bug of ours.

If I had to guess, it could be that this code already runs when the window is not open yet. I am on wxgtk3 version 3.0.4-2, and gtk3 3.24.7. All on Arch Linux.

PR #2005 (2019-03-27): Fix tree-related assertions on startup on Linux
This seems to completely fix two assertions that happen on startup (one in `Tab::update_page_tree_visibility` and one in `Tab::update_changed_tree_ui`).

Tested on a fully updated Arch Linux, running GNOME on Wayland.

Fixes #2003.
closed 2019-03-27 by vojtechkral