prusa3d/PrusaSlicer #2004: MMU2S single color pick your own extruder instead of channel 1.

Dutchy-79, 2019-03-22:
### Version
1.41.3

### Operating system type + version
Win10 64

### 3D printer brand / version + firmware version (if known)
MK3S MMU2S

### Behavior

_Is this a new feature request?_
YES

#### Project File (.3MF) where problem occurs
Not needed, New future YES

I just received the MMU2S and now I want to do a single color print, Yes I know I should print only 5 colors from now on.

Though I don need 1 color only now so I wanted red and it's in channel 3 so I choose Original Prusa i3 MK3S MMU2S Single printer te get rid of the purge block. Loaded my item. set the extruder to 3 in the model settings and when hitting export and it says: Configuration is not valid: One or more object were assigned an extruder that the printer does not have.

I want to select a channel and if technically not possible please make it so I can't select extruder 5 in the model settings.

Regards,

Jasper

sgoll, 2019-03-22:
I think with "MMU2s Single" the expected workflow is that you export the G-code with no explicit extruder selected (i.e. leave it at Extruder 1).

When you start printing, the printer should ask you which extruder to actually use for the current print.

In other words: no need to preselect the extruder during slicing but rather select it just before printing through the LCD display.

This is explained in more detail in the MMU2S Printing Handbook, chapter 9 "Printing in Single Mode" (page 29 in version 1.0.3):

https://prusa3d.com/downloads/manual/prusa3d_manual_mmu2_eng_1_03.pdf

Dutchy-79, 2019-03-22:
Thanks, I'm one that's asking or taking apart before reading ;)

closed 2019-03-22 by Dutchy-79

sgoll, 2019-03-24:
@Dutchy-79 I guess with me it's the other way around: still waiting for my MMU2S to arrive so I am stuck reading the manual and making myself familiar so that I'm all set when the unit eventually arrives. ;)