prusa3d/PrusaSlicer #2006: Support problems

espr14, 2019-03-22:
Slic3rPE 1.42.0 alpha 7 Win64

Supporting overhang test brings lot of errors:

1. Support is generated on the ceiling.
2. Support is generated for less than 55° which was set as a threshold angle.
3. Support interface starts at 70°, not 55°.

[image]

3MF project:  [overhang.zip](https://github.com/prusa3d/Slic3r/files/2998288/overhang.zip)

Kolous, 2019-03-24:
Have the same observation too. And some extra lines on the opposite side of the model.

philipr-za, 2019-08-20:
Also observing similar behaviour, lack of interface layers and little dots of support on upper surfaces. Observed in PrisaSlicer 2.0.0 and 2.1.0.

ProTubeVR, 2021-03-05:
Still have this behavior on last 2.3.0 version
[image]

ProTubeVR, 2021-03-05:
This issue is quite super painful to me...... please Help !

bubnikv, 2021-03-22:
> Support interface starts at 70°, not 55°.

That is intentional. The steep overhangs do not need the dense support.

closed 2026-09-29 by prusaslicer-bot[bot]