prusa3d/PrusaSlicer #2007: Part of second layer missing

leftspin2k, 2019-03-22:
### Version
1.42.0-beta_win64

### Operating system type + version
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 17134), 64-bit edition

### 3D printer brand / version + firmware version (if known)
Prusa MK3 stock
firmware 3.5.1 release build

### Behavior
Slice using the stock configuration for 0.3mm Draft mode

export

Print the file.

first layer prints fine, second layer does not extrude during first 20% of the layer. printer moves over the region.

Print 1 and print 2 top sides

Reproduction of the print defect.

[image]

Print 1 top side and Print 2 bottom side.

Shows that extrusion picks up at the same location in both prints. Not a filament feed issue unique to 1 machine.

[image]

#### Project File (.3MF) where problem occurs

[Draft bug.zip](https://github.com/prusa3d/Slic3r/files/2998348/Draft.bug.zip)

bubnikv, 2019-03-24:
I don't know what is happening here, but I inspected your G-code in Simplify3D (it has a g-code viewer) and I see no problem with the G-code. I am afraid I cannot help you.

leftspin2k, 2019-03-24:
OK, I will post on the firmware then.

closed 2019-04-04 by leftspin2k