prusa3d/PrusaSlicer #2008: custom g-code at tool-change is random

JohnArild, 2019-03-22:
### Version
1.42.0-beta+win64-full-201903171406

### Operating system type + version
Win10 Pro

### 3D printer brand / version + firmware version (if known)
Prusa MK3S

### Behavior
When I change number of extrudes from one to two and add M600 to tool change g-code, it only sometimes works. For some reason it changes my M600 to some MMU2 specific code in the g-code output. The filament selection and layer height settings changes to MMU2 specific selections at arbitrary points.
I have not selected MMU2 as my printer so I don't want MMU2 specific code in my generated g-code, but I do want dual extrusion with custom tool-change g-code. Sometimes it works as I intend.

bubnikv, 2019-03-24:
Try deselect the following:

[image]

JohnArild, 2019-03-25:
Thanks, that seems to work. For some reason I thought that had to be selected since I am doing multiple materials on a single extruder. But now that I think about it, the slicer won't know the difference if I don't put in any extruder offset. 
Maybe the checkbox could be more clearly marked? Although, now that I know, it can be marked whatever for what I care.

closed 2019-03-25 by JohnArild