prusa3d/PrusaSlicer #2009: Independent layer height for infill/support

espr14, 2019-03-23:
While perimeters should have low or variable layer height, infill and support should have independent layer height, typically high for high speed. The only catch is to fit precise height of perimeters, infill and support. This can be solved by setting height of the last layer of infill/support to suitable value.

bubnikv, 2019-03-24:
Supports are already printed with their own layer height.

espr14, 2019-03-24:
Where can I change support layer height?

bubnikv, 2019-03-24:
You don't. Support generator will generate as thick layers as possible, if
possible. The maximum layer height is defined at the extruder page (printer
settings).

espr14, 2019-03-24:
These are layers where support is added (for some model). Maximal layer height was set to 0.5. Average layer height of support is 0.47375 only. There was no support interface near.

6.34
6.82
7.29
7.76
8.23
8.71
9.18
9.65
10.13

Still for the infill, you can set only multiplier of the perimeter layer height. If the perimeter layer height is 0.15 and the maximal is 0.25, you have to stick with 0.15 for the infill.

bubnikv, 2019-04-13:
thank you for your suggestion.

closed 2019-04-13 by bubnikv