prusa3d/PrusaSlicer #2010: Skin

espr14, 2019-03-23:
Some types of skin can be made by CAD SW but some are easier to do in slicer. It could be another setting next to support. Skin can be created by slight changes of how the model is printed or by slight changes of model. Here are some possibilities.

1. Every seem creates a small hole. If you create more seems at the same layer in random distance, it will create doted skin. Printer can first print odd and then even parts or it can make a circle at the seem to cut the material. Otherwise, it would blend the seem.

2. Printer creates waves at the sharp corners. Let's create waves everywhere with random phase. Even frequency and amplitude can be changed continuously. This will create uneven, still smooth surface.

bubnikv, 2019-03-24:
Sorry, I understood neither of your proposals. 

1) An image would help
2) Uneven still smooth? I don't understand that.

espr14, 2019-03-24:
Issue in random seems created a doted skin:
[image]

Sinus skin:
[image]
[image]

bubnikv, 2019-03-24:
did you print that sinus skin?

espr14, 2019-03-24:
Not yet.

x0rtrunks, 2019-03-24:
So like the fuzzy skin option in Cura?

espr14, 2019-03-25:
Yes, but easier to implement and with more possibilities.

espr14, 2019-03-26:
Sinus skin feels even better than it looks:

[image]

It is less slippery than smooth surface. Changing amplitude does not make sense so just change phase of wave.

clfaye, 2019-12-06:
I would also love to see this in PrusaSlicer

NateroniPizza, 2020-01-21:
+1 to this - this would be awesome for handles for things.

szaghi, 2020-01-29:
I really like the fuzzy skin feature of CURA, although I have just started to play with it: below is my first test with CURA's fuzzy skin with *a not so rational setting*, but result is promising in my opinion. 

I hope to see fuzzy skin or alike feature in PrusaSlicer soon.

Cheers

[image]
[image]

espr14, 2020-01-29:
Still, Cura does not have fuzzy skin on top. This could be done by alternating z height in interval of layer height while compensating extrusion.

a-wigand, 2020-03-07:
+1
This would be a great feature!

qJake, 2020-07-03:
Yes! Please add Fuzzy Skin support to PrusaSlicer!

When you combine:

* The textured PEI sheet
* Fuzzy walls
* Maybe a fuzzy top layer from @espr14's idea?

Your whole model can have a fuzzy/rocky look and feel to it! This would be awesome!

JakubSido, 2020-07-16:
+1 to this - this would be awesome

konopik, 2020-07-22:
+1 please

DD8HL, 2020-08-25:
Please add fuzzy everything!

pbrgle1, 2020-09-17:
> Yes! Please add Fuzzy Skin support to PrusaSlicer!
> 
> When you combine:
> 
> * The textured PEI sheet
> * Fuzzy walls
> * Maybe a fuzzy top layer from @espr14's idea?
> 
> Your whole model can have a fuzzy/rocky look and feel to it! This would be awesome!

You can get a kind of fuzzy top layer with Hilbert Curve infill.

+1 for fuzzy perimeters!

foreachthing, 2020-09-17:
If I could make a wish:
Add a Fuzzy-Skin-Modifier. That way I can choose where the fuzz needs to be.

reddawgesq, 2020-10-12:
I REALLY need this feature/capability!

wolfgangmauer, 2020-11-03:
I REALLY need this feature/capability !!!

maciii, 2020-11-23:
+1 to add fuzzy skin

JamesLear92, 2020-12-08:
I love Prusa Slicer, but I've been using cura for lots of my prints now because of the Fuzzy Skin option. It really helps hide artifacts, and gives a nice texture a bit like the spring steel powder coated PEI print bed. 
Would love this feature.

Gertsch, 2020-12-18:
+1 pleas add this to Plicer!

Hooch180, 2020-12-30:
I too would love to see this featrue.

bubnikv, 2021-02-11:
Implemented the same way as in Cura, even using the contour fuzzification algorithm by Cura.

closed 2021-02-11 by bubnikv

wolfgangmauer, 2021-02-17:
@bubnikv Great job!
Another important option is "Fuzzy Skin Outside only" !!!

bubnikv, 2021-02-18:
> Another important option is "Fuzzy Skin Outside only" !!!

That is in there.

wolfgangmauer, 2021-02-18:
@bubnikv I just tried, almost perfect.
The first layer has no fuzzy skin !!!

bubnikv, 2021-02-19:
> The first layer has no fuzzy skin !!!

That is on purpose. Cura does the same.

wolfgangmauer, 2021-02-19:
What makes you think that? This is not the case with Cura.
[image]

Gertsch, 2021-03-02:
Is there a release with fuzzy skin or do I have to compile myself? How would I get it?

adarcher, 2021-03-20:
This or #2024 was closed prematurely.  One of them needs to be open if this hasn't been merged in yet.

JosephMarinier, 2021-04-29:
@Gertsch, @adarcher
https://github.com/prusa3d/PrusaSlicer/commit/4620402ab44af3547ddc2f9ebbeb579261df0cef appears to be merged in master and tagged with `version_2.4.0-alpha0`
Thanks a lot @bubnikv!