prusa3d/PrusaSlicer #2011: Automatic chamfers

espr14, 2019-03-23:
Chamfering can be done in CAD SW but it is fixed to 45° typically. Moreover, you have to decide on which side it will be printed on.

Slicer is deeper in the print itself so it is better to do chamfering during slicing. Advantages:
1. Slicer chamfers parts where it is necessary, not all parts.
2. Angle of chamfer can be changed to arbitrary angle (not just 45°).
3. You can rotate model and see what chamfering does. There's no need for re-rendering of the model.
4. You can decide whether use chamfer or support during slicing or combine both.

Chamfering can be done in 2 opposite ways. Either you go from bottom to top and you cut model, or you go from top to bottom and add mass to the model. Both are useful and they can be different in the same model by using modifiers.

Spacemarine2018, 2019-03-24:
What do you want to achieve with this? Reduce elephant foot?

espr14, 2019-03-24:
There's extra parameter for elephant foot and it solves first layer only. Chamfering helps handle overhangs in design.

If slicer does chamfering, you don't need to think about chamfering during model design. Moreover, printer handles 55° or even more with smaller layer height. Definitely, during design you don't know what the rotation and layer height will be. Also different material/printer can behave differently. These information is in slicer, not CAD SW.

Spacemarine2018, 2019-03-24:
Ok, so your goal is to avoid overhang angles above a certain threshold angle (like 55° or 60° depending on material and printer) by adding or removing material at the bottom of the model?

That makes sense in my eyes. The way you described it in your first post led me to belive that you wanted a general chamfering option to like turn a cylinder into a chamfered cylinder at the bottom. Hence my question what you wanted to achieve with this.

espr14, 2019-03-24:
It is about chamfering complete model so it can be printed without support. It's not just about chamfering on the bed.

jhoblitt, 2019-03-25:
If slic3r is considered the CAM equivalent of a typical "subtractive manufacturing" CAD/CAM workflow, where chamfers are often added at the CAM stage, then I suppose it would be reasonable for slic3r to support this level of object manipulation.

However, I personally would consider this an extremely low priority because:

1) virtually all CAD software has support for chamfers and I know that at least fusion 360 allows both the angle and distance to be controlled.
2) I dislike the fundamental argument of "my CAD sucks so your slicer needs to make up for it".
3) chamfering could just as easily be done post CAD with a plethora of mesh manipulation programs before being imported by a slicer.
4) I suspect automatic chamfering for all but trivial meshes will be hard and require a number of user configurable parameters to tune edge selection and even then will require part modifiers to block its action.
5) As this feature isn't wide spread, meshes made for distribution will still need to be chamfered by the creator 
6) slic3r doesn't seem to have a concept of UI edge/face selection or modifiers, so even manual edge chamfering support would require a lot of plumbing.

@espr14 Could you provide an example of software package or published algorithm that selects edges for chamfering?

supermerill, 2019-03-25:
Shouldn't be very hard to add. When slicing, only keep the polygons that are a bit larger than the ones from the previous layer. Like the support detection.

jhoblitt, 2019-03-25:
chamfering is significantly more complicated than support generation as the later doesn't need to worry about pre-existing supports and any additional geometry is not supposed to change the final created part. How is the slicer supposed to divine pre-existing chamfers?  Should a 44.9 degree external edge be chamfered? What if the edge to be chamfered has a vertex that also forms an internal corner?  My gut suspicion is there are a lot of edge cases (pun intended).

It seems like someone should at least demonstrate an external stl chamfering script that can handle more than trivial shapes, perhaps the mk3s parts, before this should be considered as a slic3r feature.

supermerill, 2019-03-25:
imo, the hard part is not chamfering the bridges.

espr14, 2019-03-25:
Algorithm is easy and does not need to process STL or think about already designed chamfers or supports. Algorithm processes single layer (A) by using previous layer (B) so 2D operations only. Previous layer could be one up or down depending on whether you want additive/subtractive chamfering.

Subtractive chamfering:
1. Find bridges in A and remove them from further (chamfer) processing.
2. Remove parts that are marked by modifier where chamfering should not be done (blocker or for additive method).
3. Inflate layer B by the width computed from user defined angle and layer height.
4. All overhanging parts of A are removed from the original layer A.

Additive chamfering:
1. Find bridges in B and add them to A.
2. Add parts to A that are marked by modifier where chamfering should not be done (blocker or for subtractive method).
3. Deflate layer B by the width computed from user defined angle and layer height.
4. All overhanging parts of B are added to original layer A.

espr14, 2019-03-25:
Do you know any freeware mesh processing SW which can do chamfering?

jhoblitt, 2019-03-25:
http://lmgtfy.com/?q=blender+chamfer

jhoblitt, 2019-03-26:
How would looking up or down one layer be able to distinguish between a sharp edge and a curve?

espr14, 2019-03-26:
Blender can chamfer but it cannot change angle and you have to do it edge by edge.
You don't need to distinguish between shapes. Or is there a reason?

AJolly, 2023-11-02:
Would love that.

closed 2026-09-29 by prusaslicer-bot[bot]