prusa3d/PrusaSlicer #2012: Filament selection not persisting

jimmy00784, 2019-03-23:
### Version
1.42 beta
### Operating system type + version
Arch Linux

### 3D printer brand / version + firmware version (if known)
Custom printer - multi-extruder single nozzle setup - Marlin 1.1.9

### Behavior
Filament Selection not saved when switching printer profiles.
Until version 1.41, filament selection persisted across printer profile changes as well as restarts.
In version 1.42, filament selection resets as soon as printer profile is changed from one printer profile to another profile.

[image]

bubnikv, 2019-03-24:
Would you please attach the content of your configuration directory or a config bundle?

Would you please retest with beta1?
https://github.com/prusa3d/Slic3r/releases/tag/version_1.42.0-beta1
Some update issues of the preset combo boxes were fixed, but I am not sure whether it will help in your case.

jimmy00784, 2019-03-24:
Beta 1 doesn't address this issue.
I've attached the config bundle as requested.
To recreate the test, try the following with the attached configuration:
1. With printer "Anet Quad" selected, set 4 different filament values. Note the new values.
2. Change printer to "AnyCubic i3 Mega". 
3. Change printer back to "Anet Quad". Note the values again. They are all same as the filament in extruder 1 and won't match the values you originally setup in step 1.

[Slic3r_config_bundle_1.42.0.b.1.ini.txt](https://github.com/prusa3d/Slic3r/files/3000723/Slic3r_config_bundle_1.42.0.b.1.ini.txt)

bubnikv, 2019-04-13:
This seems to be working in beta1. Would you please verify?

bubnikv, 2019-05-04:
it should be fixed. closing.

closed 2019-05-04 by bubnikv