prusa3d/PrusaSlicer #2013: Incorrect Filament on extruder 1 in 3D and Layer view

jimmy00784, 2019-03-23:
### Version
1.42 beta

### Operating system type + version
Arch Linux

### 3D printer brand / version + firmware version (if known)
Custom printer - multi-extruder setup - Marlin 1.1.9

### Behavior
In multi-extruder printer setup:
If you change extruder 1 filament after starting Slic3r, extruder 1 filament isn't getting applied. 
Instead, the original filament is still used. Extruder 2, 3, 4 etc work fine. 

A similar issue is seen in a single extruder printer setup.

The issue can be confirmed in the generated G-Code, if you include an identifying comment in each of the filament's custom g-code. 

[image]

bubnikv, 2019-03-24:
Would you please retest with the beta1?
https://github.com/prusa3d/Slic3r/releases/tag/version_1.42.0-beta1

Some update issues of the preset combo boxes were fixed.

jimmy00784, 2019-03-24:
I confirm. Beta 1 fixes this issue. Thanks.

bubnikv, 2019-03-24:
cool, thanks.

closed 2019-03-24 by bubnikv