prusa3d/PrusaSlicer #2014: [1.42.0b1] Support and Infill drop-down incorrectly sized on HiDPI

nebbywan, 2019-03-23:
### Version
1.42.0 beta 1 (commit 9652e8193a48d53441c2210a6f2d988b3ba811e1)

### Operating system type + version
Win10 Home x64 Build 17763

### Behavior
On a HiDPI display, the "Support" and "Infill" drop-down menu heights on the right-hand sidebar are too short. This does not seem to be an issue on a standard (1080) display.
[image]

bubnikv, 2019-03-24:
Is it a single or dual display setup? We found a similar issue on a dual
screen setup, if happen to open Slic3r on a secondary monitor, while each
monitor has a different scaling assigned.

nebbywan, 2019-03-24:
It is on a single display setup at 150% scaling

cosmith71, 2019-03-24:
I can confirm this.  2 monitor setup with different scaling, but with Slic3r on the main monitor.  HiDPI.

[image]

sufertashu, 2019-03-27:
Same issue for dropdowns on Print Settings > Infill view (and all other dropdowns in general).

YuSanka, 2019-03-27:
Thanks for your report. There is new problem, concerned with wxWidgets. 
When we don't use Bitmap for wxBitmapComboBox, default item hight is 13 (for some unknown reason). 

Working on it in [SPE-889](http://dev.prusa3d.com/browse/SPE-889)

darthvader85, 2019-03-27:
I have the same issue on a single monitor setup with a 4k monitor. Scaling in Windows is set to 200%.
[image]

FidelCapo, 2019-04-01:
This issue should be fixed in next release.
Closing.

closed 2019-04-01 by FidelCapo

nebbywan, 2019-04-01:
Confirmed as fixed in 9594bf43b5d3fd1eeaf44ac982408b2c66f4288f. Thank you @FidelCapo !

FidelCapo, 2019-04-02:
Thanks belongs to YuSanka :)