prusa3d/PrusaSlicer #2018: Send to printer fails if octoprint is using forced https

GurliGebis, 2019-03-24:
### Version
1.41.1

### Operating system type + version
Windows 10 and Octoprint running on a raspberry pi with the Octopi distro.

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3 - older firmware, not relevant for this issue.

### Behavior
If I change haproxy to require https (and using http 301 redirect to move the client from http to https), it is not possible to use the "Send to printer" button.

To reproduce, configure slic3r to be able to send gcodes directly to octoprint.
Then change the haproxy config, so it redirects http trafic to https.
Try and send a gcode to the printer, and what as nothing happens on the printer side.

As soon as the direct is disabled, it works as it always has.
The certificate used is issued by a CA that is trusted by the machine, so it shouldn't be a problem there.
Most likely, slic3r tries to upload using http, recieves the redirect, and then fails instead of trying using https.

bubnikv, 2019-03-24:
Please test with
https://github.com/prusa3d/Slic3r/releases/tag/version_1.42.0-beta1
The 1.41.3 will not be updated.

GurliGebis, 2019-04-14:
Hey, sorry for the late reply.

I just tried with 1.42.0-beta1, and it says the upload is complete, but the file does not show up in octoprint.

If SSL is forced, it says Completed immidiately, if I don't force SSL, it uploads normally, and the file shows up.

Reading in the Printer Settings pane, it says "To use a custom CA file, please import your CA file into Certificate Store / Keychain." - is this the Windows certificate store? (Because the CA is already added there, so it should be working).

Is there a way to see debug info for what the upload tries to do, and what result it gets?

GurliGebis, 2019-04-14:
Okay, if I add the "https://" in front of the hostname, it seems to be working.

Most likely related to it defaulting to http and not https, if no scheme is defined in the URL:

https://github.com/prusa3d/Slic3r/blob/44a611795b10a3773f31b688d7d9ba74095fd086/src/slic3r/Utils/OctoPrint.cpp#L173-L184

May I suggest changing it, so it tries https first if no scheme is defined, and then fallback to http and try with that instead?

vojtechkral, 2019-04-15:
@GurliGebis Right, thanks for the report, I'll look into it ...

GurliGebis, 2019-04-15:
Great, I'll change it back and test again when the next beta is available :)

vojtechkral, 2019-04-15:
You can test with the following build. Basically the redirecting should now be fixed for g-code uploading.

[Slic3rPE-1.42.0-beta2+6-vk-bugfixes-win64-full-gaf6019075-201904151140.zip](https://github.com/prusa3d/Slic3r/files/3079951/Slic3rPE-1.42.0-beta2%2B6-vk-bugfixes-win64-full-gaf6019075-201904151140.zip)

GurliGebis, 2019-04-15:
Just tested it - it works fine with just the hostname now, so it understands the http redirect now :)

I'll close the issue, since it has been resolved.

closed 2019-04-15 by GurliGebis

vojtechkral, 2019-04-16:
Great, thanks for testing!

lbussy, 2020-03-19:
Just in case someone finds this and is having the same issue I was:

- If you changed your hostname away from 'octopi' you need to [generate a new certificate](https://github.com/PrusaMK2Users/MK2_Tips_and_Tricks/wiki/Generating-a-new-SSL-certificate-for-OctoPi) with the new name
- You need to use the server name, not the IP address which will populate if you browse to the server
- You need to prepend the server name with `https://`
- Do not use the `'local` portion of the name

I have Bonjour installed (Windows) so I know my system can resolve the .local addresses.  I don't have a machine to test on without Bonjour.  If you cannot browse to {servername}.local with a web browser, you might need to install Bonjour to make it all work.

bubnikv, 2020-03-19:
thanks for comment, I will link it to a new issue opened recently. I would be thankful if you can help there.

brombomb, 2021-06-03:
> Just in case someone finds this and is having the same issue I was:
>
> * If you changed your hostname away from 'octopi' you need to [generate a new [...]

This helped me immensely.  Since I print everything from one machine (linux) I was able to edit my hosts file to point octopi directly to the IP without the `.local` and that resolved all my connection issues.

lbussy, 2022-11-15:
> How to proceed? How to import the self signed cert into Windows so PrusaSlicer accepts this?

This is not a Prusa question; this is a Google question:

https://learn.microsoft.com/en-us/biztalk/adapters-and-accelerators/accelerator-swift/adding-certificates-to-the-certificates-store-on-the-client

commit 554176b9 (2019-04-16): Http: Fix redirecting of post requests, fix #2018