prusa3d/PrusaSlicer #2019: [Feature request] dynamic z lift for travel

burner-, 2019-03-24:
Currently z axis of travel are linear or you can add z lift to travel. Also z lift with wipe is possible. With travel at plane without zlift with high speed can cause damage for printer, head or print when it hits to blobs. With large prints & nozzles it is some times quite rough ride when extruder travel long distances without z lift. Z lift helps for that but it slow down prints and with large nozzles it cause even more blobs and random failures to prints because when you hit 200-300mm/s to blob with 1.2mm nozzle it is quite hard collision. So solution for that would be make dynamic parabolic shape z lift. It can work by calculating time for xy acceleration from print speed to max travel speed and calculating z acceleration so that lift will take same time than zy acceleration but it is made simultaneously with xy travel.  Because x&y need time to accelerate it will give soft lift ramp at same time. When x/y speed is increased to dangerously high also z is already lifted to safe distance. When travel start slow down also z will start come back to print plane. That way z lift wont make blobs and it will not slow down print at all and it will reduce failure possibility with high travel speeds.

burner-, 2021-01-29:
Someone suggest "airplane travel" as better name for this :D

Jan-Soustruznik, 2025-03-19:
Closed as solved. The requested feature has already been implemented in Slicer version 2.7.2-Alpha2 as **Ramping lift** located in Printers → Extruder

for more info check the Changelog: [version_2.7.2-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.2-alpha2)

Thank you, and have a great day!

closed 2025-03-19 by Jan-Soustruznik